volumio/volumio-plugins #17: Documentation?

paviro, 2016-08-13:
Is there any documentation available? How would I for example execute a function whenever audio starts and stops to play?

balbuze, 2016-08-21:
Now, have a look here  : https://volumio.github.io/docs/Plugin_System/Plugin_System_Overview.html

paviro, 2016-08-21:
I tried writing one but I always get `The folowing error occurred when installing the plugin: TypeError: Cannot read property 'name' of undefined` even when installing the official spotify plugin.

buguibu, 2016-10-15:
I faced the same error and my problems were: 

- i was making the *.zip* of the folder instead of all the files and folder
- do not forget to add the _node_modules_ folder to the plugin zip
- as said in [The plugins.json file](https://volumio.github.io/docs/Plugin_System/Plugin_System_Overview.html) i forgot to add my plugin to _/data/plugins/plugins.json_

closed 2017-04-21 by crisp00