volumio/volumio-plugins #19: Volume control not working playing songs from Spotify plugin with I2S DAC

mueckinger, 2016-08-21:
Volume control works fine playing webradio and I2S DAC.
When using I2S DAC switching to Spotify songs via Spotify plugin, volume is always 100%.
When switching to Raspi Audio Output there is no problem with volume control.

Audio Output: I2S with phatDAC (=HifiBerry DAC) with soft volume mixer using this tweak:
https://volumio.org/forum/rc2-volume-options-mixer-available-t4318.html#p20194

balbuze, 2016-08-21:
It can't work with that as Spotify plugins does not use mpd and thus changing mpd.conf has no effect for that.
You have to change  also /etc/spopd.conf output_name = softvol (not sure, to be checked)
But a software volume control is coming...

volumio, 2016-08-21:
Exactly, this is not a bug per se. We will add software volume control

closed 2016-08-21 by volumio

mueckinger, 2016-08-21:
changing
/etc/spopd.conf output_name = softvol 
worked
THX!

rmcmahon, 2016-12-18:
Excellent, that was my issue too, it was set to hardware. I'm using the hifiberry amp+, and a rpi2b but had the same issues, i.e. web radio working but spotify not playing sound. thanks.

trgraglia, 2017-05-17:
This fix does not work any longer for me with release 2.175, released 16.05.2017. Can anyone else confirm?