volumio/volumio-plugins #27: Plugin - Touch Display requiring action

kaicheen, 2016-10-15:
`Downloading plugin at http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/miscellanea/touch_display/touchdisplay.zip
END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/miscellanea/touch_display/touchdisplay.zip
Creating folder on disk
Unpacking plugin
Checking for duplicate plugin
Copying Plugin into location
Installing dependencies
Installing Chromium Dependencies
Hit http://archive.raspbian.org jessie InRelease
Hit http://archive.raspbian.org jessie/main Sources
Hit http://archive.raspbian.org jessie/contrib Sources
Hit http://archive.raspbian.org jessie/non-free Sources
Hit http://archive.raspbian.org jessie/main armhf Packages
Hit http://archive.raspbian.org jessie/contrib armhf Packages
Hit http://archive.raspbian.org jessie/non-free armhf Packages
Ign http://archive.raspbian.org jessie/contrib Translation-en
Ign http://archive.raspbian.org jessie/main Translation-en
Ign http://archive.raspbian.org jessie/non-free Translation-en
Installing Graphical environment
Download Chromium
Install Chromium
Selecting previously unselected package chromium-browser.
(Reading database ... 24459 files and directories currently installed.)
Preparing to unpack .../chromium-browser_48.0.2564.82-0ubuntu0.15.04.1.1193_armhf.deb ...
Unpacking chromium-browser (48.0.2564.82-0ubuntu0.15.04.1.1193) ...
Selecting previously unselected package chromium-codecs-ffmpeg-extra.
Preparing to unpack .../chromium-codecs-ffmpeg-extra_48.0.2564.82-0ubuntu0.15.04.1.1193_armhf.deb ...
Unpacking chromium-codecs-ffmpeg-extra (48.0.2564.82-0ubuntu0.15.04.1.1193) ...
Setting up chromium-codecs-ffmpeg-extra (48.0.2564.82-0ubuntu0.15.04.1.1193) ...
Processing triggers for mime-support (3.58) ...
Processing triggers for man-db (2.7.0.2-5) ...
Reading package lists...
Building dependency tree...
Reading state information...
Correcting dependencies... Done
The following extra packages will be installed:
gconf-service gconf2-common libauthen-sasl-perl libencode-locale-perl
libfile-basedir-perl libfile-desktopentry-perl libfile-listing-perl
libfile-mimeinfo-perl libfont-afm-perl libgconf-2-4 libgnome-keyring-common
libgnome-keyring0 libhtml-form-perl libhtml-format-perl libhtml-parser-perl
libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl
libhttp-daemon-perl libhttp-date-perl libhttp-message-perl
libhttp-negotiate-perl libio-html-perl libio-socket-ssl-perl
liblwp-mediatypes-perl liblwp-protocol-https-perl libmailtools-perl
libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl
libnspr4 libnss3 libtie-ixhash-perl libtimedate-perl liburi-perl libwww-perl
libwww-robotrules-perl libx11-protocol-perl libxml-parser-perl
libxml-twig-perl libxml-xpathengine-perl libxss1 xdg-utils
Suggested packages:
libdigest-hmac-perl libgssapi-perl gnome-keyring libdata-dump-perl
libcrypt-ssleay-perl libauthen-ntlm-perl libunicode-map8-perl
libunicode-string-perl xml-twig-tools gvfs-bin
The following NEW packages will be installed:
gconf-service gconf2-common libauthen-sasl-perl libencode-locale-perl
libfile-basedir-perl libfile-desktopentry-perl libfile-listing-perl
libfile-mimeinfo-perl libfont-afm-perl libgconf-2-4 libgnome-keyring-common
libgnome-keyring0 libhtml-form-perl libhtml-format-perl libhtml-parser-perl
libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl
libhttp-daemon-perl libhttp-date-perl libhttp-message-perl
libhttp-negotiate-perl libio-html-perl libio-socket-ssl-perl
liblwp-mediatypes-perl liblwp-protocol-https-perl libmailtools-perl
libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl
libnspr4 libnss3 libtie-ixhash-perl libtimedate-perl liburi-perl libwww-perl
libwww-robotrules-perl libx11-protocol-perl libxml-parser-perl
libxml-twig-perl libxml-xpathengine-perl libxss1 xdg-utils
0 upgraded, 44 newly installed, 0 to remove and 0 not upgraded.
97 not fully installed or removed.
Need to get 5559 kB of archives.
After this operation, 18.5 MB of additional disk space will be used.
Get:1 http://archive.raspbian.org/raspbian/ jessie/main gconf2-common all 3.2.6-3 [1040 kB]
[... 202 lines cut ...]
Please select the layout matching the keyboard for this machine.

1. English (US)
2. English (US) - Cherokee
3. English (US) - English (Colemak)
4. English (US) - English (Dvorak alternative international no dead keys)
5. English (US) - English (Dvorak)
6. English (US) - English (Dvorak, international with dead keys)
7. English (US) - English (Macintosh)
8. English (US) - English (US, alternative international)
9. English (US) - English (US, international with dead keys)
10. English (US) - English (US, with euro on 5)
11. English (US) - English (Workman)
12. English (US) - English (Workman, international with dead keys)
13. English (US) - English (classic Dvorak)
14. English (US) - English (international AltGr dead keys)
15. English (US) - English (left handed Dvorak)
16. English (US) - English (programmer Dvorak)
17. English (US) - English (right handed Dvorak)
18. English (US) - English (the divide/multiply keys toggle the layout)
19. English (US) - Russian (US, phonetic)
20. English (US) - Serbo-Croatian (US)
21. Other`

The installation of the plugin Touchscreen gets stuck here.

volumio, 2016-10-15:
Please update to latest volumio version (2.001) , then it will install just fine

kaicheen, 2016-10-15:
Works! Thanks :)

raesche, 2016-10-28:
I have the same issue. I have downloaded and installed volumio version 2.001 2016-10-15 and the problem remains.

Any ideas how to solve this issue? is there a way to manually install the plugin so I can select the keyboard-configuration through the console?

Thanks & best regards

[image]

Alphascrypt, 2016-11-06:
Hello, 
i think you are using the odroid C2 with the volumio 2.002 image. Still reboot and open up the console,
just type     sudo dpkg --configure -a     .

After this install the plugin ;)

closed 2017-06-01 by volumio

AntonK123, 2019-03-31:
I have the same issue too. Alphascrypt's method is not working for me.

uiblogit, 2024-12-16:
Fresh install of latest volumio on la raspberry Pi 4B, i have this error when try to install this plugin for use my touch hdmi display:

Downloading plugin at https://plugins.volumio.workers.dev/pluginsv2/downloadLatestStable/touch_display/volumio/buster/armhf
END DOWNLOAD: https://plugins.volumio.workers.dev/pluginsv2/downloadLatestStable/touch_display/volumio/buster/armhf
Creando la destinazione
Estraendo il plugin
Verifica che il plugin sia adatto a questa versione di Volumio The plugin can be used with this version of Volumio
Controllando se vi sono plugin duplicati
Copiando il plugin nella destinazione
Installando i servizi accessori
Re-synchronizing package index files from their sources
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 https://deb.nodesource.com/node_14.x buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Ign:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [110 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian buster/rpi armhf Packages [1360 B]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [18.3 MB]
Ign:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [69.3 MB]
Fetched 69.5 MB in 1min 58s (588 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Installing graphical environment
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
keyboard-configuration libegl-mesa0 libegl1 libegl1-mesa libepoxy0 libevdev2
libfontenc1 libgbm1 libgudev-1.0-0 libinput-bin libinput10 libmtdev1
libpciaccess0 libunwind8 libutempter0 libwacom-bin libwacom-common libwacom2
libwayland-server0 libxaw7 libxcomposite1 libxfont2 libxft2 libxkbfile1
libxmu6 libxmuu1 libxpm4 libxt6 libxxf86dga1 x11-utils x11-xkb-utils xauth
xbitmaps xfonts-base xfonts-encodings xfonts-utils xserver-common
xserver-xorg xserver-xorg-core xserver-xorg-input-all
xserver-xorg-input-libinput xserver-xorg-input-wacom xserver-xorg-legacy
xserver-xorg-video-all xserver-xorg-video-amdgpu xserver-xorg-video-ati
xserver-xorg-video-fbdev xserver-xorg-video-nouveau
xserver-xorg-video-radeon xserver-xorg-video-vesa xterm
Suggested packages:
pciutils mesa-utils xfonts-100dpi | xfonts-75dpi xfonts-scalable xinput
firmware-amd-graphics xserver-xorg-video-r128 xserver-xorg-video-mach64
xfonts-cyrillic
The following NEW packages will be installed:
keyboard-configuration libegl-mesa0 libegl1 libegl1-mesa libepoxy0 libevdev2
libfontenc1 libgbm1 libgudev-1.0-0 libinput-bin libinput10 libmtdev1
libpciaccess0 libunwind8 libutempter0 libwacom-bin libwacom-common libwacom2
libwayland-server0 libxaw7 libxcomposite1 libxfont2 libxft2 libxkbfile1
libxmu6 libxmuu1 libxpm4 libxt6 libxxf86dga1 x11-utils x11-xkb-utils xauth
xbitmaps xfonts-base xfonts-encodings xfonts-utils xinit xserver-common
xserver-xorg xserver-xorg-core xserver-xorg-input-all
xserver-xorg-input-libinput xserver-xorg-input-wacom xserver-xorg-legacy
xserver-xorg-video-all xserver-xorg-video-amdgpu xserver-xorg-video-ati
xserver-xorg-video-fbdev xserver-xorg-video-nouveau
xserver-xorg-video-radeon xserver-xorg-video-vesa xterm
0 upgraded, 52 newly installed, 0 to remove and 2 not upgraded.
Need to get 19.0 MB of archives.
After this operation, 32.1 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libgbm1 armhf 19.3.2-1~bpo10+1~rpt4 [67.2 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf libegl-mesa0 armhf 19.3.2-1~bpo10+1~rpt4 [120 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf libegl1 armhf 1.3.0-7~bpo10+1 [28.5 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf libegl1-mesa armhf 19.3.2-1~bpo10+1~rpt4 [49.4 kB]
Get:7 http://archive.raspberrypi.org/debian buster/main armhf xserver-common all 2:1.20.4-1+rpt4+deb10u9 [2236 kB]
Get:2 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf keyboard-configuration all 1.193~deb10u1 [404 kB]
Get:9 http://archive.raspberrypi.org/debian buster/main armhf xserver-xorg-core armhf 2:1.20.4-1+rpt4+deb10u9 [3281 kB]
Get:5 http://raspbian.mirror.garr.it/mirrors/raspbian/raspbian buster/main armhf libxxf86dga1 armhf 2:1.1.4-1 [22.4 kB]
[... 419 lines cut ...]
Purging configuration files for xserver-xorg-core (2:1.20.4-1+rpt4+deb10u9) ...
rm: cannot remove '/var/log/Xorg.*.log': No such file or directory
rm: cannot remove '/var/log/Xorg.*.log.old': No such file or directory
Purging configuration files for libxxf86dga1:armhf (2:1.1.4-1) ...
Purging configuration files for xfonts-utils (1:7.7+6) ...
dpkg: warning: while removing xfonts-utils, directory '/usr/share/fonts/X11/util' not empty so not removed
Purging configuration files for libfontenc1:armhf (1:1.1.3-1) ...
Purging configuration files for xfonts-base (1:1.0.5) ...
dpkg: warning: while removing xfonts-base, directory '/usr/share/fonts/X11/misc' not empty so not removed
Purging configuration files for x11-utils (7.7+4) ...
Purging configuration files for xterm (344-1+deb10u2) ...
[... cut ...] 10u1) ...
Purging configuration files for xserver-xorg-legacy (2:1.20.4-1+rpt4+deb10u9) ...
Processing triggers for fontconfig (2.13.1-2) ...
Deleting /opt/volumiokiosk.sh
Deleting /data/volumiokiosk
Deleting /data/volumiokioskextensions
Deleting /lib/systemd/system/volumio-kiosk.service
Enabling login prompt
Done
Errore durante l'installazione del plugin Error
Removing plugin directory /data/plugins/user_interface/touch_display
Removing temporary directory /tmp/touch_display-63HZOksvMi

[image]