volumio/volumio-plugins #32: Spotify no sound with 2.001 on rpi3

snowf0x, 2016-10-22:
Hey guys, 
When I try to play a Spotify song, webui show notification music start to play, but play/stop button is not refreshed. However the playing progress circle is showing and time is counting as well.
Most important no sound is come out.  I use iquadlO dac. Playing nas music and webradio is working fine.

PaulFinch, 2016-11-09:
I think I have the same behavior with my Phat Dac, and I think it could come from a specific setting in
/etc/spopd.conf : output_name = ${outdev} resulting in hw:0 or hw:1 depending on the default one I guess ... Sometime, it could be the unused one.

In my case, I don't want to use this setting because I'm using a the softvol thing to control the volume (PhatDac doesn't have hardware volume control) so I get the sound but not "mixed"

PaulFinch, 2016-11-10:
Actually, volumio is quite sensitive when changing playback settings.
However, i managed to get it work by changing /etc/spopd.conf. Like I said i use the softvolume device. But I had to change output_name = hw:softvolume by output_name = softvolume

ashthespy, 2016-11-17:
I seem to have the same issue
Hardware: pi0 + PCM5102 I2C DAC ('Hifiberry') with no hardware volume control.
I can play files off my NAS no issues, the software volume works fine. However, from Spotify, I have the same issues - No sound.
I did try what @PaulFinch  suggested, and changed `/etc/spopf.conf` tried `output_name = softvolume`  and 'output_name = softvol' but no luck. Any ideas?

tabroughton, 2017-04-16:
I have similar issue tested on rpi2 and rpi3 with Pi-DAC+, playback settings set to hardware. Spotify plugin has no sound and I've had to force output to hardware by changing `output = hw:softvolume`  to `output = hw:0` unsure why spotify plugin isn't picking up that the mixer type is set to hardware in gui.

tabroughton, 2017-04-17:
Have just done volumio factory reset to try to replicate and it worked perfectly.  1) set up pi-DAC+ playback settings, 2) ssh into volumio and set both analog settings in alsamixer for IQaudioIO to 100% (as recommended in IQaudIO docs), 3) installed spotify plugin. output is set to HW:1 (card 0 is still onboard bcm chip) and all is working perfectly... 

possible that the plugin gets confused when changing playback settings whilst installed which is what I may have had issues with when first time.

closed 2017-05-08 by volumio