volumio/volumio-plugins #43: Spotify plugin: not possible to install with changed user password

pagnotta, 2017-01-29:
After having changed the default password for the user `volumio` it is not possible to install the spotify plugin anymore. What I get in `journalctl` is

```
Jan 29 10:41:20 volumio volumio[919]: info: Downloading plugin at http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/spotify/spotify.zip
Jan 29 10:41:21 volumio volumio[919]: info: END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/spotify/spotify.zip
Jan 29 10:41:24 volumio volumio[919]: info: Checking if plugin already exists
Jan 29 10:41:24 volumio volumio[919]: info: Rename folder
Jan 29 10:41:25 volumio volumio[919]: info: Move to category
Jan 29 10:41:25 volumio volumio[919]: info: Checking if install.sh is present
Jan 29 10:41:25 volumio volumio[919]: info: Executing install.sh
Jan 29 10:41:25 volumio sudo[1135]: pam_unix(sudo:auth): authentication failure; logname= uid=1000 euid=0 tty= ruser=volumio rhost=  user=volumio
Jan 29 10:41:28 volumio sudo[1135]: pam_unix(sudo:auth): conversation failed
Jan 29 10:41:28 volumio sudo[1135]: pam_unix(sudo:auth): auth could not identify password for [volumio]
Jan 29 10:41:28 volumio sudo[1135]: volumio : 1 incorrect password attempt ; TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/sh /data/plugins//music_service/spop/install.sh
Jan 29 10:41:28 volumio volumio[919]: [sudo] password for volumio: Sorry, try again.
Jan 29 10:41:28 volumio volumio[919]: [sudo] password for volumio:
Jan 29 10:41:28 volumio volumio[919]: sudo: 1 incorrect password attempt
Jan 29 10:41:28 volumio volumio[919]: info: Install script return the error Error: Command failed: echo volumio | sudo -S sh /data/plugins//music_service/spop/install.sh > /tmp/installog
Jan 29 10:41:28 volumio volumio[919]: [sudo] password for volumio: Sorry, try again.
Jan 29 10:41:28 volumio volumio[919]: [sudo] password for volumio:
Jan 29 10:41:28 volumio volumio[919]: sudo: 1 incorrect password attempt
Jan 29 10:41:28 volumio volumio[919]: info: An error occurred installing the plugin. Rolling back config
```

When trying to revert to the old password using `passwd` it is rejected since it is "too simple".

So I had to do `sudo passwd volumio` to revert to the old password. Then installation of the spotify plugin worked.

marianol, 2017-06-10:
+1 on this issue. 

Had to do the same to install the plugin. As a possible fic you can have the UI ask for the password on install

fanthore, 2017-06-19:
Same issue with Youtube plugin install

crisp00, 2017-06-20:
Aww snap, I'll look into that

WoBBeLnl, 2017-06-26:
Any news on this issue? Is there any place this "volumio" password is hardcoded?

Simple work around: change password back to "volumio"

dhoffend, 2017-06-26:
Yes the password is hardcoded in the volumio2 pluginmanager and many other places :-/

See this issue
https://github.com/volumio/Volumio2/issues/1192

Examples:
https://github.com/volumio/Volumio2/blob/f160a0f3a61804e759f9475c6694f03ba1928bc8/app/plugins/system_controller/volumio_command_line_client/volumio.sh#L55
https://github.com/volumio/Volumio2/blob/4ade247cea43b22c6f11160bf11804dee9c04e52/app/plugins/system_controller/volumio_command_line_client/commands/kernelsource.sh#L16
https://github.com/volumio/Volumio2/blob/c21785b0877305c02ee26eb8fce7c92443d19e4c/app/pluginmanager.js#L873

dhoffend, 2017-06-26:
One workaround would be disallow the "volumio" user to connect via SSH (sshd_config). Then you can keep your default password and be secure.

Disable SSH via WebUI is useless if you can enable it again without any form of authentication

volumio, 2017-06-27:
You know that this is a great idea!!! We could create a different volumio user (say volumiod) to handle all the installations that require sudo privileges...