volumio/volumio-plugins #44: Spotify plugin: "My Playlists" not opening

pagnotta, 2017-01-29:
When going into `Browse > Spotify` and clicking on "My Playlists" folder, nothing happens. In `journalctl` I get

```
Jan 29 11:03:37 volumio volumio[919]: info: CoreCommandRouter::executeOnPlugin: spop , handleBrowseUri
Jan 29 11:03:37 volumio volumio[919]: info: [1485687817455] ControllerSpop::sendSpopCommand
Jan 29 11:03:37 volumio volumio[919]: info: ADDING DEFER FOR COMMAND ls
Jan 29 11:03:37 volumio volumio[919]: info: FIRST BRANCH
Jan 29 11:03:37 volumio volumio[919]: info: BEFORE: SPOP HAS 1 PROMISE IN STACK
Jan 29 11:03:37 volumio volumio[919]: info: AFTER: SPOP HAS 0 PROMISE IN STACK
Jan 29 11:03:37 volumio volumio[919]: info: FIRST BRANCH
Jan 29 11:03:37 volumio volumio[919]: info: BEFORE: SPOP HAS 0 PROMISE IN STACK
Jan 29 11:03:37 volumio volumio[919]: info: AFTER: SPOP HAS 0 PROMISE IN STACK
```

The other folder as e.g. "What's New" are working though.

Playing "My Playlists" was working until maybe two or three weeks ago. So I guess it might have to do with a Spotify API change?

Also a reinstall of the system did not change the situation.

pagnotta, 2017-01-29:
By the way, I am on Odroid-C2 Volumio v2.03, Spotify plugin version from 13-10-2016.

pagnotta, 2017-01-29:
After having a look at this report in the forum:

https://volumio.org/forum/large-spotify-playlists-not-working-t5348.html

I deleted some Spotify Playlists, and it works again!

So the problem is too many playlists on Spotify. Seems like the Spotify plugin is not able to handle those.

maracuja-juice, 2018-02-09:
@pagnotta How many playlists did you have in the end?

pagnotta, 2018-02-10:
After more than one year that has passed, I can only give you a rough estimate: I'd say about somewhere between 50 to 150?

maracuja-juice, 2018-02-10:
@pagnotta Okay. When I tried this out I had the same issue but I only have about 9 playlists. 50 is a bit more than that :)

imitto, 2019-02-28:
I have the same issue today. It worked fine in the morning, then a playlist endd and I can not start a new one from "my playlists". The playlists are there, but it can't add them to cue or even can not open them to see the tracks in them. Any other thing (my save albums, my top artists etc.) wors fine.
- I tried to disable-enable plugin, it didn't help.
- Tried to uninstall the plugin but that cannot be done. There plugin remains in the installed plugins folder, with the "Enable" button in half way(!!!) If I try to uninstall anyother plugin, it's the same. I tried it from an another PC, if I navigate to the installed plugins folder, I see the same as on the other PC - so it's not just some cached stuff.
- I reinstalled Volumio from the basics. Fresh image, fresh install, same problem.
- Generated a new refresh token, edited index.js, uploaded it, restarted, same problem.