volumio/volumio-plugins #54: Raspberry Pi doesnt boot (panic) after adding touchscreen plugin

markcarline, 2017-03-09:
volumio works fine on the raspberry PI 3 (not sure if it's all Pis) but after adding the official touch screen display plug in it gives a Kernel panic, see:

[image]

iamsi, 2017-03-09:
I have the touch working fine on a pi3 how do you have the touch screen connected, do you have the I2C pins connected on the gpio header for example?

markcarline, 2017-03-09:
Yes, but they are mapped via an IQAudio PI-DigiAmp+ (it's the same pins that are used).

Everything worked yesterday fine, i just re-imaged from as new download today (which seems to use a new IMG file dated 2017-03-07)

iamsi, 2017-03-09:
remove the i2c pins, if you are using a pi3 it should come over the display flex cable.

markcarline, 2017-03-09:
Yep tried that (it's only using the gnd & power and still panic)

iamsi, 2017-03-09:
I just flashed the image from yesterday (volumio-2.118-2017-03-07-pi) and I can replicate the issue. Sorry for wasting your time with the checks, the old image but with the latest updates is what I was using when I first commented.

macmpi, 2017-03-10:
Please check ongoing debug work [here](https://volumio.org/forum/installing-the-touch-display-plugin-t6032-10.html#p29746).
Not very far from being done I think: just a X.org config issue now, any help appreciated.

macmpi, 2017-03-12:
Fixed with newer plugin version: you may close this issue. Thanks.

markcarline, 2017-03-12:
Nice one guys - Just reimaged using "volumio-2.118-2017-03-07-pi.img" and added the plugin and everything works great.  thanks!

Only thing I noticed is that when adding the plug in the date was still:

"Updated: 15-10-2016"

Cheers.

iamsi, 2017-03-13:
confirm this works for me too. Thanks!

macmpi, 2017-03-30:
We can certainly close this one now. Thanks.

closed 2017-03-30 by volumio