volumio/volumio-plugins #60: Touch screen : no gui

MaitreDede, 2017-04-12:
Hi, I just installed volomio (volumio-2.129-2017-03-23-pi) on my Pi3 with touchscreen.

I installed the Touch Display plugin.

When I activated it, only mouse is shown, no gui at all.

caveman-dick, 2017-05-02:
I'm getting the same issue on 2.141, did you manage to resolve this?

jackson15j, 2017-12-29:
Just installed a new Pi3 with touch screen this morning and no longer seeing this issue:

* `dd` Volumio 2.246 (2017-07-31 build) to a microsd card.
* Configured Volumio to use local Wifi and then rebooted.
* Installed touch screen plugin.
* Activated touch screen plugin.
* Mouse cursor shown on screen.
* **Not timed!!** ~30secs the Volumio UI is displayed on the screen.

Might be worth trying on an updated image to see if this is still an issue for you?