volumio/volumio-plugins #77: Plugin installation fails when volumio password has changed

dhoffend, 2017-06-23:
I've changed to volumio users password to make my secure my player in the network. But when I tried to install the youtube plugin the installation fails cause it tries to execute a command using sudo and gets forced to enter a password which then fails

```
2017-06-23T21:53:51.025Z - info: Downloading plugin at http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/youtube/youtube.zip
2017-06-23T21:53:55.424Z - info: END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/youtube/youtube.zip
2017-06-23T21:54:09.506Z - info: Checking if plugin already exists
2017-06-23T21:54:09.511Z - info: Rename folder
2017-06-23T21:54:12.881Z - info: Move to category
2017-06-23T21:54:19.947Z - info: Checking if install.sh is present
2017-06-23T21:54:19.954Z - info: Executing install.sh
2017-06-23T21:54:22.813Z - info: Install script return the error Error: Command failed: echo volumio | sudo -S sh /data/plugins//music_service/youtube/install.sh > /tmp/installog
[sudo] password for volumio: Sorry, try again.
[sudo] password for volumio:
sudo: 1 incorrect password attempt

2017-06-23T21:54:22.825Z - info: An error occurred installing the plugin. Rolling back config
2017-06-23T21:54:22.826Z - info: Plugin folders cleanup
```

I would prefer storing correct sudo rules in `/etc/sudoers.d/` rather then piping a default password into the install script. Please check your install.sh method.

P.S. I know that the newest version of volumio disables ssh per default but that really doesn't secure the system as long as you can enable the ssh daemon via UI without any form of authentication.

WoBBeLnl, 2017-06-26:
I think this is the same as https://github.com/volumio/volumio-plugins/issues/43

dhoffend, 2017-06-26:
It's not a plugin problem, it's hardcoded in volumio2 code
https://github.com/volumio/Volumio2/issues/1192

closed 2017-06-26 by dhoffend