volumio/volumio-plugins #85: cannot activate auto_play plugin

tomzu, 2017-07-29:
I tried the auto_play plugin. I was able to install it, but I cannot switch it active. If I try, the browser changes to the playback page and the waiting icon appears for a while. When it disappears, nothing has happend, i.e. when I return to the plugins page, auto_play is still marked as inactive.
I am running Volumio 2.201 on a Raspi 3 B, and I am using the current auto_play from https://github.com/volumio/Volumio2-UI/files/761286/auto_play.zip

tomzu, 2017-07-29:
Update: I just upgraded to Volumio 2.245, to no avail.

volumio, 2017-07-29:
Contact plugin developer

tomzu, 2017-07-31:
I thought this page was the way to contact the developer? @thomasdoerr

thomasdoerr, 2017-07-31:
Hi tomzu, 

you need to address the developer like you did it in your last post, otherwise I don't get an email.

Can you send me the logs?
Here you can find, how to find them: https://volumio.github.io/docs/Development_How_To/Debugging.html

I'm running the plugin on two raspberry pi 3 since I developed it. They are both now running on version 2.201, one with HiFiBerry and the other with a "china import" sound card. 

I'll update one of my machines to the nevest version of volumio and let you know if it still works as expected.

Regards,
Thomas

thomasdoerr, 2017-07-31:
Both machines are on version 2.224 and on both the plugin is working as before.
I'm unable to get the newest version as you mentioned, I don't know why, but both are not reacting on a button click on "check updates" :-(
Would be nice, if you could send some logs.
Thomas

volumio, 2017-07-31:
@thomasdoerr see here:
https://volumio.org/forum/update-224-225-t7192.html

ssh then sudo ldconfig -v

tomzu, 2017-07-31:
@thomasdoerr Thanky you. I'll see what I can do, but I'm definitely not able to install a development environment on a Mac, as it is proposed in the video on the web page you mentioned above.
However, I did a journalctl -f, and this doesn't look quite good.
[autoplaylog.txt](https://github.com/volumio/volumio-plugins/files/1188568/autoplaylog.txt)
I was wondering what the problem with that title was, as that isn't currently played, nor the next in the playlist, but can be played without a problem.

thomasdoerr, 2017-07-31:
thanks @volumio I just read this in this post :-) it works fine! Both are downloading the new version.

thomasdoerr, 2017-07-31:
@tomzu is "Title: Be OK" something you see in your play queue? I'm using my plugin with a web radio and as far as I remember with spotify, but probably not with titles of music you stored on your disk. Perhaps you could clean your queue and try it again?

tomzu, 2017-07-31:
@thomasdoerr Well ... actually cleaning the queue or removing all the playlists and favourites that contain that title is not sufficient. Only if I UNMOUNT the NAS storage, I can activate the plugin. If I remount the NAS, starting the plugin will fail again. (Btw. stopping the plugin results in a red error message box popping up.) 
And an even more interesting thing: If I JUST remove that title from my NAS, the plugin will also work with the NAS mounted!
So, what is so wrong with "Ingrid Michaelson - Be OK.mp3"?

tomzu, 2017-07-31:
@thomasdoerr Obviously, something in the system thinks that a title containing "OK" is not okay. I renamed that song to "Ingrid Michaelson - Be okay.mp3", and it works ... strange.
Nevertheless, the error popup on plugin deactivation remains; this is of course less important, just irritating.

thomasdoerr, 2017-07-31:
@tomzu I recently tried another lib in my plugin, just to find the issues with Daniels volumio instance.
You can find our discussion here: https://github.com/volumio/Volumio2-UI/issues/219

I gave him a tweaked version of my plugin: 
https://github.com/volumio/Volumio2-UI/files/811426/auto_play.zip

You could also try it. 
When I now read the thread I mentioned above, I noticed that there was a simular problem.

Hope that helps
Regards,
Thomas

andrew-james-heard, 2018-02-18:
@thomasdoerr I have an identical experience to @tomzu with previous Volumio 2.348 & latest 2.362. Today I've done a complete fresh install of Volumio on an RPi3B, and get exactly the same error with the Autoplay plugin - perform install; when enabled volumio appears to completely restart and then checking the "installed plugins" shows that Autoplay is again disabled. This occurs each time. So I perform sudo journalctl -f then attempt to enable the plugin then save the console output. Should I post the whole 225 lines here? Here below are the first lines of volumio log; I think it contains the error:
``Feb 19 07:49:55 volumio volumio[953]: info: Enabling plugin auto_play
Feb 19 07:49:55 volumio volumio[953]: info: Loading plugin "auto_play"...
Feb 19 07:49:55 volumio volumio[953]: info: ControllerAutoPlay - connecting mpd on host: localhost; port: 6600
Feb 19 07:49:55 volumio volumio[953]: info: PLUGIN START: auto_play
Feb 19 07:49:55 volumio volumio[953]: info: Done.
Feb 19 07:49:55 volumio volumio[953]: /data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:189
Feb 19 07:49:55 volumio volumio[953]: if(err) { throw err; }
Feb 19 07:49:55 volumio volumio[953]: ^
Feb 19 07:49:55 volumio volumio[953]: Error: Non okay return status: "Title: I'm OK"
Feb 19 07:49:55 volumio volumio[953]: at MPD._checkReturn (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:353:10)
Feb 19 07:49:55 volumio volumio[953]: at MPD.<anonymous> (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:188:18)
Feb 19 07:49:55 volumio volumio[953]: at MPD._handleResponse (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:450:3)
Feb 19 07:49:55 volumio volumio[953]: at MPD._onData (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:344:10)
Feb 19 07:49:55 volumio volumio[953]: at emitOne (events.js:101:20)
Feb 19 07:49:55 volumio volumio[953]: at Socket.emit (events.js:188:7)
Feb 19 07:49:55 volumio volumio[953]: at readableAddChunk (_stream_readable.js:176:18)
Feb 19 07:49:55 volumio volumio[953]: at Socket.Readable.push (_stream_readable.js:134:10)
Feb 19 07:49:55 volumio volumio[953]: at TCP.onread (net.js:547:20)
Feb 19 07:49:55 volumio volumio-remote-updater[605]: [2018-02-19 07:49:55] [error] handle_read_frame error: websocketpp.transport:7 (End of File)
Feb 19 07:49:55 volumio volumio-remote-updater[605]: [2018-02-19 07:49:55] [disconnect] Disconnect close local:[1006,End of File] remote:[1006]
Feb 19 07:49:55 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Feb 19 07:49:55 volumio systemd[1]: Unit volumio.service entered failed state.
``

thomasdoerr, 2018-02-19:
Hi Andrew,
Unfortunately I never had this error here on my maschine. So I was never able to reproduce it.
The others fix it by renaming the file (which must be something like "I'm OK.mp3" or so in your case) and then it worked. Perhaps there is an encoding "problem" in the name of the file. I don't know. Could you try to do that?
Regards,
Thomas

andrew-james-heard, 2018-02-20:
@thomasdoerr thanks but I can't find any file "*ok.mp3"? So maybe that is part of the problem? Below was the Volumio install "details":
`Downloading plugin at http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/miscellanea/auto_play/auto_play.zip
END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/miscellanea/auto_play/auto_play.zip
Creating folder on disk
Unpacking plugin
Checking for duplicate plugin
Copying Plugin into location
Installing dependencies
Adding plugin to registry
Plugin Successfully Installed
`

thomasdoerr, 2018-02-20:
Can you check what happens when you clear your playlist?
In this thread we discussed that. https://github.com/volumio/Volumio2-UI/issues/219

andrew-james-heard, 2018-02-20:
I don't have any playlist. Nothing to clear sorry. I do have items in the queue. This problem occurred on a fresh install: installed Volumio, installed Touch Display, installed Autoplay.

thomasdoerr, 2018-02-20:
I'm sorry Andrew, I meant the queue, not a playlist. 
The plugin plays the first entry of the queue.

andrew-james-heard, 2018-02-20:
Thanks for helping Thomas. I cleared the queue then enabled Autoplay, Volumio restarted but Autoplay is still disabled. I used sudo journalctl –f while enabling but same result as before:

Feb 20 19:45:49 volumio volumio[2601]: /data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:189
Feb 20 19:45:49 volumio volumio[2601]: if(err) { throw err; }
Feb 20 19:45:49 volumio volumio[2601]: ^
Feb 20 19:45:49 volumio volumio[2601]: Error: Non okay return status: "Title: I'm OK"
Feb 20 19:45:49 volumio volumio[2601]: at MPD._checkReturn (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:353:10)
Feb 20 19:45:49 volumio volumio[2601]: at MPD.<anonymous> (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:188:18)
Feb 20 19:45:49 volumio volumio[2601]: at MPD._handleResponse (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:450:3)
Feb 20 19:45:49 volumio volumio[2601]: at MPD._onData (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:344:10)
Feb 20 19:45:49 volumio volumio[2601]: at emitOne (events.js:101:20)
Feb 20 19:45:49 volumio volumio[2601]: at Socket.emit (events.js:188:7)
Feb 20 19:45:49 volumio volumio[2601]: at readableAddChunk (_stream_readable.js:176:18)
Feb 20 19:45:49 volumio volumio[2601]: at Socket.Readable.push (_stream_readable.js:134:10)
Feb 20 19:45:49 volumio volumio[2601]: at TCP.onread (net.js:547:20)
Feb 20 19:45:49 volumio volumio-remote-updater[606]: [2018-02-20 19:45:49] [error] handle_read_frame error: websocketpp.transport:7 (End of File)
Feb 20 19:45:49 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Feb 20 19:45:50 volumio volumio-remote-updater[606]: [2018-02-20 19:45:49] [disconnect] Disconnect close local:[1006,End of File] remote:[1006]
Feb 20 19:45:50 volumio systemd[1]: Unit volumio.service entered failed state.

thomasdoerr, 2018-02-21:
Are you using a network drive with media files?
I never did that, so probably in this area there is still a problem.

What happens if you play something, so the queue is filled and then you try to enable autoplay?
Then it should find your first entry in the queue.

andrew-james-heard, 2018-02-21:
No, just using a USB flash drive. I see exactly the same stack trace with 0 or 1 or many tracks in the queue and playing or paused.

thomasdoerr, 2018-02-24:
Hi Andrew, I tried to reproduce your problem on my machine. Added a USB stick with music, put that music into the queue, delete everything from the stick (with another machine, so that volumio doesn't know, that songs have been deleted).
But it doesn't crash. I noticed, that booting with a 64GB USB stick seems to take to long, so that my plugin isn't starting the last song, but volumio works without any problem. I used version 2.368.
Did you try to boot without your USB stick?

andrew-james-heard, 2018-02-24:
Thanks Thomas. My RPi doesn't boot any slower with 64GB USB stick.

I enabled the plugin with tracks still in queue - same log Error: Non okay return status: "Title: I'm OK"
I deleted all tracks from queue then enabled the plugin - same error

I say again my install is very fresh & few changes to default settings. Could I attempt to debug the error? Add any extra print/ debug statements?

Fybrox, 2018-02-25:
Hello and sorry for my poor english, i'm french.
I have problem with this plugin too, (last Volumio and plugin releases on Pi3).
With only webradios in queue and no usb drive or nas connected it run well as expected.
When i boot with my music files on usb drive, volumio bootloop on "ding ding ding" music intro.
When i desactive the plugin, usb disk connected, volumio boot well but no autoplay of course. 
This is my experience, i'm not a "power user" so it will be difficult for me to help you on what append in my case, sorry.
I hope you find a solution because this plugin is a very good idea. Thanks for the work.

thomasdoerr, 2018-02-25:
Thanks for your comment and thanks for using my plugin, even if it doesn't work in all situations :-) . I'll try to debug that stuff, but I can't exactly tell you when I'm able to do that. I'll keep you posted.
Bon weekend à tous!

andrew-james-heard, 2018-02-25:
Here is another topic with my experience: https://volumio.org/forum/plug-autoplay-dont-work-volumio-version-246-t7654.html#p45483.

chiu0602, 2018-03-11:
I am using RPI3 and met this issue, I SSH into volumio and found that when line 50 of auto_play/index.js is commented, volumio can boot, which the line is
`mpd.connect();`

then I wrote a little code and found that when calling `mpd.connect();`, following error is shown:
```
/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:189
                if(err) { throw err; }
                          ^

Error: Non okay return status: "Title: BOK"
    at MPD._checkReturn (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:353:10)
    at MPD.<anonymous> (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:188:18)
    at MPD._handleResponse (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:450:3)
    at MPD._onData (/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js:344:10)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at TCP.onread (net.js:547:20)
```

Look into the MPD module which the plugin using, there is a function `_checkReturn(msg)` check with the `msg` content. Exception will be thrown when `msg !== "OK"`.
So I just changed `/data/plugins/miscellanea/auto_play/node_modules/node-mpd/src/mpd.js` line 352
from
`if(msg !== "OK") {`
to
`if(!msg.includes("OK")) {`

And the plugin work as expected.

Actually, have to fix "Title: BOK" is returned as status instead of modifying the node module code.

thomasdoerr, 2018-03-11:
Hi, thanks for your debugging!

do you see the same behaviour with the version I changed last yearr to use mpd.js instead of node-mpd.js?
You can still find it here: https://github.com/thomasdoerr/volumio-plugins/tree/mpd-instead-of-node-mpd/plugins/miscellanea/auto_play
Regards,
Thomas

chiu0602, 2018-03-11:
@thomasdoerr Your new version works perfectly! Thanks for your plugin, many people need it. Please merge as soon as you can =)

andrew-james-heard, 2018-03-17:
I just did a complete fresh install of image 2.382 onto an RPi3B with MP3 files on USB drive (no change from previous reports). Enabling the Autoplay plugin still causes volumio to restart & I observe Autoplay is still disabled (no change from previous reports).

thomasdoerr, 2018-03-17:
thanks for reporting! Since chiu0602 wrote, that the autp_play version with mpd.js works fine, I'll port the new library, test it on my machines and commit it. Hope that this fixes your problem too.
https://github.com/volumio/volumio-plugins/issues/85#issuecomment-372118846

andrew-james-heard, 2018-03-17:
I will be looking forward to testing.

dyno2400, 2018-03-18:
Hello.
I want to use this plugin. It is launched and works.
I use an USB. When playing the list of tracks, when restarting playing begins with the first track in the list.
Why not from that track which was lost at the time of switching off of Pi3?
I use Pi3 B, Volumio 2.378
Thanks.

thomasdoerr, 2018-03-18:
Hi, 

I implemented the plugin, because I needed a way to start playing my web radio when I turn it on. I never implemented a storage, which saves the latest state somewhere. My PI is also never shutdown properly, instead I'm unplugging it hard. So there would never be an event for the plugin to store the state. Instead it should do that always while a track changes or periodically. This is much more complex to implement and that's why it is not included. Feel free to extend it!
Regards,
Thomas

thomasdoerr, 2018-03-20:
@andrew-james-heard @dyno2400 @Fybrox @tomzu @chiu0602 

Good morning folks, I published a new, rewritten plugin which is based on another mpd lib. Some of of you already tested a version.
In order to provide you a smooth migration, I called the new plugin "AutoStart". It will replace auto_play. But before I remove auto_play, I'd like you to move over to the new one and give me feedback. 

Please DO NOT enable autostart and auto_play at the same time! If the new one works for you, remove the old one.

You can search for the new plugin in volumio plugins under Miscellanea. 

[image]

I'm looking forward to getting your feedback!

dyno2400, 2018-03-20:
When playing the list of music with an USB, the new plug-in works as well as old. Everything is good.

chiu0602, 2018-03-22:
It works perfectly! Thanks @thomasdoerr

andrew-james-heard, 2018-03-23:
Work perfectly for me too! Tested with single queued webradio, single queued USB track, and multiple queued USB tracks. Thanks very much @thomasdoerr.

Fybrox, 2018-03-24:
Hello guys, many thanks to chiu0602 and thomasdoerr "autostart" work fine now in all situations, very good job :))

andrew-james-heard, 2018-03-24:
Would it be equally possible to AutoStart with last item in the queue rather than first? It should not require storing any state. The last item is not necessarily the same as last item played, but it would be more likely than first queued item. I have done Python programming but not JavaScript like in the plugin. Is this idea possible and where should I look to investigate further?

houndofsound, 2018-03-24:
Hi Thomas 
is there any prospect of this plugin moving to a 'restart last played' basis rather than rerunning the play queue from the beginning?
I would guess that volumio does not keep track of the last played member of the playlist, only the playlist itself, and that might be why your plugin cannot address this desirable feature....

andrew-james-heard, 2018-03-25:
@houndofsound - note the items are from the queue, not playlist. But last item would still be an improvement on 1st item, and should not require "tracking" I think.

chiu0602, 2018-03-25:
The plugin just perform "press play button" on Volumio start, so the plugin have done its work already. This plugin should not implement additional features.

This is Volumio's current "feature" that it play first audio instead of the last playing audio. The code on Volumio's core should be changed to fix this "feature" 

I used a workaround that enabled shuffle play, and put a silent audio as the first item on my playlist.

hessercan, 2021-12-27:
Does AutoStart work on the Pandora Plugin? I need a way for my store to resume playback of a pandora station after a power outage.

dyno2400, 2021-12-27:
Hello. I don't use Pandora.

thomasdoerr, 2021-12-28:
Hi, sorry same with me, I don’t use it.
Best, Thomas