volumio/volumio-plugins #91: Youtube plugin search does not seem to work at all

sorin-pop, 2017-08-22:
Youtube plugin search does not seem to work at all

sla89, 2017-10-02:
Can you please give more details about your setup and volumio version? I have it running on raspberry pi 3 with latest volumio and everything works fine.

sla89, 2018-04-24:
@sorin-pop friendly ping

sorin-pop, 2018-04-25:
Hi, I'm sorry for the late reply. I have the same setup and I think last time I used it it was working fine. Anyway it's not important for me anymore, as I got myself a second raspberry pi 3 with LibreElec+Kodi+Youtube addon for Kodi, so I am rather using Youtube from that one now. Still using the Volumio Pi for only-audio stuff though (web radios, spotify, etc.).

Thanks for your assistance. I am closing the ticket as per the above.

closed 2018-04-25 by sorin-pop