volumio/volumio-plugins #92: Youtube plugin setting Code unclear how to use

sorin-pop, 2017-08-22:
Not clear what to do.

Enter the following code on {0}. What does {0} mean?

I get a code of 10, but when I try to enter it at https://www.google.com/device I cannot type it in.

This really seems like very very buggy and unprofessionial. Why don't you guys make this plugin payable, but then at least make it properly. So frustrating...

volumio, 2017-08-22:
@sorin-pop this plugin is made by volunteers in their free time

Why don't you grow up?

sorin-pop, 2017-08-22:
Yes, that was exactly my point as well (Why don't YOU grow up?). But it seems you did not understand my point. Too bad.

volumio, 2017-08-22:
@sorin-pop if you can't understand that this plugin is made by people (not part of Volumio Team) during their free time, and that plugins can be improved by anyone (you included) since they are open source, then there is no point getting your whinings. 
If you can comment on a constructive basis, fine. If you can commit code to make this plugin better, great. If not, just go somewhere else.

crisp00, 2017-08-23:
> Why don't you guys make this plugin payable, but then at least make it properly.

Sir, now that's the kind of ideas I love hearing!
cristian@pintea.net is my email, if you need some tips you can ask! At this point I'm waiting to see your plugin when it comes out!

closed 2017-08-23 by crisp00

sorin-pop, 2017-08-23:
OK, after a Volumio restart the Youtube plugin seems to work much better. The issues reported above are no longer present. So it's not as bad as I thought, I just needed a Volumio restart after installing the plugin. Perhaps this will be useful info for somebody else who will run into this problem.

I apologize for being so frustrated initially, and for my tone. Well, Cris, I guess I won't have to write to you directly for the moment. (although I don't really understand you mentioning some plugin of mine! That's kind of weird...This ticket was about _your_ plugin. But I'm glad you like the idea of possibly making it payable and elevating it from just a simple hobby buggy project to a more professional one)