volumio/volumio-plugins #109: No sound with spotify connect

Anders186, 2017-10-20:
Hi guys, Using volumio release 2.296, rpi3, generic I2S DAC (my own board). Working great with other media but no sound using spotify plugin. It have tried many different settings via web GUI but no luck. Use softvolume. I have tried to change the spopf.conf output_name = softvol but still no sound. 

/Anders

balbuze, 2017-10-21:
something not clear, are you talking about spotify plugin or about spotify connect plugin (volspotconnect or volspotconnect2)?

Anders186, 2017-10-21:
Its the spotify plugin for volumio 2 thats available from the web gui. Does it make sense or how do I find the info you need?

Anders186, 2017-10-21:
I just noticed there is an new 1.3.4 version :-) Its available as a zip file via the web gui spotify plugin update but there is an update error.
"Downloading fromhttp://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/spop/spop.zip
END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/spop/spop.zip
Creating folder on disk
The following error occurred when installing the plugin: undefined"

volumio, 2017-10-21:
Thanks to balbuze's discovery I fixed this problem yesterday.
If you can update the spotify plugin, if not just uninstall it and install it again. Save the spotify settings and it will just work

Anders186, 2017-10-21:
The uninstall and then install works fine. Unfortunatly the sound problem remains with the new 1.3.4 version. The difference now is however that the sound in the audio jack disappeared when I2S is selected. The sound to the audio jack works when its explicitly selected and the I2S is off. So still no sound from I2S using softvolume and spotify plugin (v1.3.4). Sound from other sources still works via I2S and softvolume.

balbuze, 2017-10-21:
I think you misunderstand one thing : you can't output on two device in same time. If you enable i2s dac, only the i2s dac will output (of course if everything is working), but no jacks. If you select jacks, you have to disable i2s...For the softvolume it weird.

Anders186, 2017-10-21:
ok. Thanks. However, the problem remains. I2S selected and Generic I2S DAC as output and as chosen DAC model. Should I try a complete fresh install or is there anything I can do to help to find the problem?

balbuze, 2017-10-21:
I fired up my rpi b with a hifiberry like dac. Let me some time to test.

balbuze, 2017-10-21:
The result : no problem, it works with my hifiberry like and softvol... Please, once you have configured volumio, before enablig spotify, test with a local track, test volume control. Once it is ok just enable the plugin, enter your credentials and save. If your credentials are already filled, save in any way!

Anders186, 2017-10-21:
I'm getting crazy about this... I did a complete new install from SD-card (2.296) and followed your steps. But no sound at I2S using Hifiberry nor Generic I2S DAC. Both these works when playing from USB-memory. What can I do or check more? Any config files for sound routing to play with?

balbuze, 2017-10-21:
!!!! The last thing I see, it's sample rate problem... A test you can do is to use the equalizer plugin (just to check if it solve your problem). The plugin operate a resampling. But it not compatible with softvol... But you'll have a idea if the problem comes from here. https://github.com/balbuze/volumio-plugins/raw/master/plugins/miscellanea/volsimpleequal/volsimpleequal.zip

balbuze, 2017-10-21:
But I'm stuck for tonight..

Anders186, 2017-10-21:
Thanks for all your effort this far!! :-)

balbuze, 2017-10-21:
Last thing, you can try spotify connect with one of the two plugin volspotconnect or volspotconnect2. Volspotconnect displays album art in the webUi. Both are compatible with softvol..

Anders186, 2017-10-21:
Wow - it works now :-) uses the EQ plugin as you suggested. No softvolume but great sound from spotify :-)