volumio/volumio-plugins #112: Spotify incomplete album listing

TomasBouska, 2017-11-09:
Hello,

When I open an album in Spotify, I get only the first 50 tracks of it. I couldn't find any setting to increase this limit. Could this be fixed please?

Thank you.

Cheers, Tomas

sweah82, 2020-02-10:
Hello,

Same here for me. It is not possible to play tracks with numbers over 50.
Has anybody a solution for this Problem?

Thank you in advance
Cheers, Chris

PoloB12, 2020-02-20:
This is a limitation on the spotify api itself :

[https://developer.spotify.com/documentation/web-api/reference/library/check-users-saved-tracks/](url)
[https://developer.spotify.com/documentation/web-api/reference/personalization/get-users-top-artists-and-tracks/](url)

mallardtheduck, 2021-01-29:
Yes, clients are supposed to use multiple requests to retrieve the complete list. Don't blame Spotify for this.