volumio/volumio-plugins #119: playback saved to local file?

lachlan-00, 2017-12-22:
Hey, I want to cache my playback with at least just the filepath to a local text file.

Is that possible with a plugin?

lachlan-00, 2018-03-14:
i have taken the last-100 plugin and tried writing to the data folder using fs-extra.

problem is that it crashes the web interface until i disable it.
does a system plugin have extra features over a user plugin?

volumio, 2018-03-14:
No, you should check your code better.
Let me see what you've done

lachlan-00, 2018-03-14:
I have the plugin creating the txt file on start so I'm  nearly there. I just cut out the last 100 i didnt want, then added fs-extra lines.

Here is a zip from the cli
https://cloud.lachlandewaard.org/index.php/s/ffM5PFqMJsjCeYc

volumio, 2018-03-14:
You're doing it wrong.
Use the plugin helper to create a new plugin:
https://volumio.github.io/docs/Plugin_System/Writing_A_Plugin.html

lachlan-00, 2018-03-14:
Will do. I'm usually a python person but I'll sit down after work and sort this out.

closed 2018-03-14 by lachlan-00

lachlan-00, 2018-03-16:
[cache_fm.zip](https://github.com/volumio/volumio-plugins/files/1818605/cache_fm.zip)

I give up. i've tried making separate functions and tried to work out how to stop the plugin reloading every time i start if but i can't work out what i'm doing.

the functions run but the web interface resets whenever i enable socket-io. I want to use sockets to get the playing song data but i've tried every existing plugin/get/push/etc and nothing works.

checkCacheruns and then it looks like it crashes at io.connection in checkPlayback.