volumio/volumio-plugins #138: Youtube plugin not retrieving stream

fighne, 2018-02-09:
Latest Volumio, using PiFi Digi+ with latest YouTube plugin not retrieving audio stream

joker5689, 2018-02-14:
Hello, i having the same problem on my Raspberry Pi 3 B, HiFiBerry AMP2 with volumio 2.362
Some YouTube songs are playing others not. I already asked for help on the volumio forum, without success. See [https://volumio.org/forum/some-youtube-songs-are-playing-others-not-t8932.html]

joker5689, 2018-02-14:
Further research showed the YouTube songs that playing have the tag "Album: null".
[image]

mxjeff, 2018-02-18:
Same thing for me on rpi2 with volumio v2.348.

I also have an issue on rpi3 with kodi + youtube plugin, [cf. related issue](/jdf76/plugin.video.youtube/issues/285).

tsrubar, 2018-02-18:
Same for me on RPi 1B+ with PiFi DAC

volumio, 2018-02-18:
Fixed with https://github.com/volumio/volumio-plugins/commit/a9e848f0dbd700b8b3976372d01b279acb2f4c48

closed 2018-02-18 by volumio