volumio/volumio-plugins #153: [volspotconnect2 0.6.0] Can not connect from mobile app

messismore, 2018-03-02:
Hi,
connecting from the macOS and Windows desktop apps works fine, but on iOS or Android it will say _connecting…_ for a while before it eventually fails. I had previously used the version from @balbuze's [repo](https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2) and it had been rock solid…

messismore, 2018-03-02:
Not sure if this is related, but I just noticed that the player shows up twice in the Spotify app: 

[image]

The player's name is _Kueche_

ashthespy, 2018-03-02:
https://github.com/balbuze/volumio-plugins/issues/133#issuecomment-369896611
Cheers! :-)

messismore, 2018-03-02:
Thanks, I had missed that 😄 Do plugins installed via the UI update automatically? I would just wait it out then…

ashthespy, 2018-03-04:
Yes and no - you should get an 'update' button when the plugin has been officially added to the plugin manager.

dudedaoud, 2018-03-06:
Il have the same problem. No connections with Android

ashthespy, 2018-03-06:
@dudedaoud  what version of the plugin are you running?

dudedaoud, 2018-03-07:
The Last 18.02.2018 plugin

messismore, 2018-03-07:
The plugin has still not been updated, unfortunately...

balbuze, 2018-03-07:
You have to install the Dev version waiting the official update.

dudedaoud, 2018-03-08:
THX but how can i download it an install ? my volumio dont have an option to install downloaded plugin.

NB : im french and i speak like a spanish cow.... 👍

ashthespy, 2018-03-08:
@dudedaoud See my previous [comment](https://github.com/volumio/volumio-plugins/issues/153#issuecomment-369936898) ;-)
> balbuze#133 (comment)
> Cheers! :-)

messismore, 2018-03-08:
Is there an ETA on this @volumio?

volumio, 2018-03-08:
Yep, testing as we speak

dudedaoud, 2018-03-08:
Great !! It work perfectly. Updated now

dudedaoud, 2018-03-08:
Vous faites un super boulot. Merci à tous les mecs !!

dudedaoud, 2018-03-10:
Hi. Dont work after update. Cant play on volumio by my Android app. Spotify updated his app against hack of apps i can see.... I have a premium account

balbuze, 2018-03-10:
Really they did ? I see an update is available on my smartphone... Maybe I shouldn't update... I'm going to check with the current version.

balbuze, 2018-03-10:
I updated spotify app to the last on my smartphone and it still working with the plugin.