volumio/volumio-plugins #163: [youtube] API no more available

camandel, 2018-03-10:
volumio can't connect to Google API for youtube service.

version: volumio v2.368 + youtube plugin
hardware: RPi3
error log:
```
2018-03-10T08:35:20.366Z - error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
Error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
    at Request._callback (/data/plugins/music_service/youtube/node_modules/google-auth-library/lib/transporters.js:85:15)
    at Request.self.callback (/data/plugins/music_service/youtube/node_modules/request/request.js:186:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1163:10)
    at emitOne (events.js:96:13)
    at Request.emit (events.js:188:7)
    at IncomingMessage.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1085:12)
    at IncomingMessage.g (events.js:292:16)
    at emitNone (events.js:91:20)
    at IncomingMessage.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)
2018-03-10T08:35:21.684Z - info: Authorization pending, polling again in 5 seconds.
```

buzuck, 2018-03-10:
I confirm the Youtube plugin stopped working since the v2.368.

I'll try to grab some logs from my RPi3-Volumio'd.

sla89, 2018-04-24:
The currently used API key is deprecated/broken and @volumio  is already informed and is initializing an API key that is managed by the volumio team itself. 

As workaround you can grant volumio access to your Youtube account. You can always revoke this!

vasyugan, 2019-08-11:
Still not fixed? It's been a year. I see in volumio.log:

```
2019-08-11T11:32:33.868Z - error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
Error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
    at Request._callback (/data/plugins/music_service/youtube/node_modules/google-auth-library/lib/transporters.js:85:15)
    at Request.self.callback (/data/plugins/music_service/youtube/node_modules/request/request.js:185:22)
    at emitTwo (events.js:126:13)
    at Request.emit (events.js:214:7)
    at Request.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1161:10)
    at emitOne (events.js:116:13)
    at Request.emit (events.js:211:7)
    at IncomingMessage.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1083:12)
    at Object.onceWrapper (events.js:313:30)
    at emitNone (events.js:111:20)
    at IncomingMessage.emit (events.js:208:7)
    at endReadableNT (_stream_readable.js:1064:12)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)

```

vasyugan, 2019-08-11:
> The currently used API key is deprecated/broken and @volumio is already informed and is initializing an API key that is managed by the volumio team itself.

Are there any news on this? We have over a year later now.

> As workaround you can grant volumio access to your Youtube account. You can always revoke this!

How do I do this? I thought what happens during pairing is precisely that  volumio is given access to the account. It says so during activation. Only that it seems to have no effect.