volumio/volumio-plugins #165: Cannot install TuneIn plugin

piffio, 2018-03-13:
When I try to install the (awesome!) TuneIn plugin via the web UI, I get the following error:

```
Mar 13 17:14:42 living-room volumio[961]: info: Downloading plugin at http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/tunein_radio/tunein_radio.zip
Mar 13 17:14:46 living-room volumio[961]: info: END DOWNLOAD: http://volumio.github.io/volumio-plugins/plugins/volumio/armhf/music_service/tunein_radio/tunein_radio.zip
Mar 13 17:14:46 living-room volumio[961]: info: Folder /tmp/plugins removed
Mar 13 17:14:47 living-room volumio[961]: ERROR: Error: stdout maxBuffer exceeded
Mar 13 17:14:47 living-room volumio[961]: info: An error occurred installing the plugin. Rolling back config
Mar 13 17:14:47 living-room volumio[961]: info: Plugin folders cleanup
```

This happens on a Raspi3.
I need to investigate the cause for this, since it seems to be a memory limit problem in the plugin manager code, something totally new to me.

I am just wondering if anyone has successfully managed to install the plugin since it has been released (I have the same repeatable issue on 2 separate Raspi3 running 2.368

balbuze, 2018-03-13:
it should be fxed with the last version of volumio 2.376

piffio, 2018-03-13:
Yes! Thanks, I had missed the latest update!

closed 2018-03-13 by piffio

volumio, 2018-03-13:
@piffio I had the issue while first installing your plugin (don't know why only your plugin triggers it), and I fixed it with https://github.com/volumio/Volumio2/commit/d6ba6a5205a35a504c3ab5b52829e07fc49e9737

So, update your Volumio to latest version and you'll be able to install it

balbuze, 2018-03-13:
@piffio , @volumio  A big thank you for this plugin ! I can now listen my local radio so easily! Well done !