volumio/volumio-plugins #169: UI not installing the latest version of the IR Remote Controller plugin

gvolt, 2018-03-18:
The plugin manager does not install the latest version (1.2.0) of the IR remote controller plugin but version 1.0.0.

After the installation the UI shows an update as of 14.11.2017. But performing the update does not install the new version either. Instead version 1.0.0 and also the indicator that an update is available remains.

It seems the ir.zip file the plugin manager downloads just has not been updated to version 1.2.0.

ashthespy, 2018-03-18:
Yes, a few plugins don't have match zips to what the repo reports - https://github.com/volumio/volumio-plugins/issues/160

sendalonline, 2018-03-21:
hi! any chance to manually force the update? thanks!

gvolt, 2018-03-22:
You can install the current version manually.

You have to connect via ssh to volumio and download the files of the latest version from [https://github.com/volumio/volumio-plugins/blob/master/plugins/accessory/ir_controller](url).

Then from the directory where you dowloaded the files

`volumio plugin install`

If an older version of the plugin is already installed I would uninstall that first. But maybe uninstalling is not necessary and "volumio plugin refresh" instead of "install" would be sufficient.

sendalonline, 2018-03-22:
Hi gvolt, thanks for your reply!

I managed to do a manuall isntall with volumio plugin install > yes, older version needs to be uninstalled. 

Unfortunately there seems to be a bug:
newer version installs the plugin in **/data/plugins/accessories/** instead of **data/plugins/system_controller/**

Volumio for some reason doesn't recognise it there. Also, upon reboot it gets deleted. 

manually moving the plugin to system_controller makes it disappear form volumio plugins. no idea what's going on.

Also, even in the latest version the conf file for the Apple remote 1156 is wrong.

Only way I got this particular remote working was installing the old plugin, and manually copying the (correct) conf files to configurations folder.

There are other bugs too but will report only after I made sure all the rest looks right.

Thanks!

gvolt, 2018-03-22:
For testing I just uninstalled and installed version 1.2.0. Installation went into "/data/plugins/**accessory**/ir_controller".

While this is the reason why `volumio plugin refresh` does not work it is also expected behaviour as one of the last changes to plugin here on github was the change of the plugin category from "system_controller" to "accessory".

If you move the plugin files manually from "accessory" to "system_controller" you need to change "endpoint":"accessory/ir_controller" to "endpoint":"system_controller/ir_controller" in line 11 of  "UIConfig.json" and probably the plugin type in "package.json" back to "system_controller".

In my case - without any changes to the plugin files or location - Volumio recognized the plugin and I had no problems changing the remote control type. Also a reboot did not delete the plugin.

closed 2019-05-13 by gvolt