volumio/volumio-plugins #178: Spotify will not transmit to bluetooth headphones like other streams.

roycenich, 2018-03-29:
Hello, I installed this popular plugin to enable bluetooth headphones:

https://volumio.org/forum/bluetooth-speaker-plugin-t7432.html

and it works great for any stream except for Spotify. I only hear Spotify through the headphone jack.

This isn't really an issue with your plugin per say, but I was thinking maybe you know why the audio output to a bluetooth sink might be different than say internet radio?

balbuze, 2018-03-30:
Try to disable/enable volspotconnect2 in order to reconfigure the plugin.

roycenich, 2018-03-30:
Thanks for the tip balbuze, I really appreciate your work.  

I gave that a try with volspotconnect2 and the other Spotify pluign but didn't have any luck. Then I tried uninstalling and reinstalling them but that didn't help either. 

Here is the plugin I'm trying to get working, maybe you might know what the problem is?

```
#!/bin/bash

echo "Installing Bluetooth Dependencies"
sudo apt-get update
sudo apt-get install -y libasound2-dev dh-autoreconf libortp-dev bluez pi-bluetooth bluez-tools libbluetooth-dev libusb-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev libsbc1 libsbc-dev

echo "Cloning Bluez-Alsa repo"
cd /tmp
git clone https://github.com/Arkq/bluez-alsa.git

echo "Building Bluez-Alsa"
cd bluez-alsa
autoreconf --install
mkdir build && cd build
../configure --disable-hcitop --with-alsaplugindir=/usr/lib/arm-linux-gnueabihf/alsa-lib
make 

echo "Installing Bluez-Alsa"
sudo make install

cat > /lib/systemd/system/bluezalsa.service <<EOC
[Unit]
Description=BluezAlsa proxy
Requires=bluetooth.service
After=bluetooth.service
[Service]
Type=simple
User=root
Group=audio
ExecStart=/usr/bin/bluealsa --disable-hfp
[Install]
WantedBy=multi-user.target
EOC

sudo systemctl daemon-reload
sudo systemctl enable bluezalsa.service

#requred to end the plugin install
echo "plugininstallend"
```

ashthespy, 2018-03-30:
copy paste from https://github.com/volumio/volumio-plugins/issues/178 for better tracking 

>That doesn't provide the info - but from the [bluez-alsa configuration guide](https://github.com/Arkq/bluez-alsa#configuration--usage) I'd reckon you might find with `cat ~/.asoundrc`.
I couldn't find the git of bluetooth-speaker-plugin, else I could have helped more..
>
>EDIT: try [`bluealsa`](https://github.com/geeks-r-us/volumio-plugins/blob/8ef0177cc180136d6b206fa9d2e3f0a5eb42e7d2/plugins/audio_interface/bluetooth_controller/index.ts#L243)

>balbuze:
>>I had a quick look at the bluetooth plugin. It seems to not configure the volumio output, but only mpd config. Two way : installing the last dev version of volspotconnect2 ad modify bye hand startconnect.sh or contact the author of the bluetooth plugin in order to make it more "standard"
>
>@balbuze  I completely agree with this, that is motivation of using a complete system like Volumio, and not manually installing mpd + librespot + gpio+ lcd + blah + blah ;-) 
>
>@roycenich  I'd urge you to open up an issue in the authors git about changing the default source. I think most of balbuze's `audio_interface`  plugins do this already, if I am not mistaken.