volumio/volumio-plugins #181: Youtube plugin account settings cannot be edited after initial setup

ViktorBarzin, 2018-04-01:
Hi all,

I've been using the youtube plugin(Updated 18.02.2018) lately and it is all okay, however, I cannot see an option to change the account settings.
I tried removing completely the plugin, unfortunately on reinstallation it appears to remember the previous settings.
Grepping the filesystem also didn't yield any results :/
So it is either the dodgy angular messing up the UI once again or the plugin keeps its settings somewhere hidden for me :/ Any ideas?

Thanks

gvolt, 2018-04-01:
I do not have a Volumio system at hand ATM, so I cannot verify. But I assume that the file or files holding your settings of the plugin should be in "/data/configuration/music_service/youtube".

ViktorBarzin, 2018-04-01:
This folder contains a config.json, however, it does not contain any information suhc as api key, email or anything like that.

gvolt, 2018-04-02:
Hm, with my RasPi still not in reach I installed Volumio (version 2.387, x86) into a VM and installed the Youtube plugin manually (version 0.0.8 from GitHub). I have only the config.json in "/data/configuration/music_service/youtube" and that indeed does not hold any account information. Since I have no Youtube account (and don't want to create one) I unfortunately cannot test "in reality".

sla89, 2018-04-24:
The plugin stores the account settings in the `authToken.json` located in `/data/plugins/music_service/youtube`. Maybe this is the wrong location for the file.

When the plugin is uninstalled volumio should clean up the whole plugin directory. That was what I have supposed it to work.

I agree, currently there is no way to remove the account settings. That is a bug. :/

To reset your account you can simply create an empty json file.