volumio/volumio-plugins #183: Use mpd instead of omxplayer for youtube plugin

volumio, 2018-04-03:
https://github.com/volumio/volumio-plugins/blob/85bd112e81062f397ef7454c90cee41b69fb47d7/plugins/music_service/youtube/index.js#L17

@sla89 if possible, let's drop omxplayer in the YT plugin and use mpd instead, so we can then release this plugin also for x86

sla89, 2018-04-04:
Sorry but I have not introduced it. It was added in  [this commit by ChrisPanda](https://github.com/volumio/volumio-plugins/commit/bd59981c0810641235a2158916116a48a3c3d438#diff-ccd38d1a25ae3876bf73f36df4d58811) and you have [already reverted it](https://github.com/volumio/volumio-plugins/commit/fb3aab898a59e084efd189ad14f913a4c18a1e12).

Removing it will break the support for omx player.

sla89, 2018-04-05:
@volumio  is it possible that it already works out of the box? See [this comment in the forum](https://volumio.org/forum/youtube-for-volumio-t6608-100.html#p47206)