volumio/volumio-plugins #185: volspotconnect - UI commands (Pause, Next, Prev, Seek) are disabled

ashthespy, 2018-04-07:
Plugin version: 0.7.0
Volumio version: 2.389

Just upgraded my Volumio, and was surprised to see that I can no longer skip songs with Spotify Connect from the WebUI, was surprised further cause this isn't visible on the dev version of the plugin?

ashthespy, 2018-04-07:
Okay, just found https://github.com/volumio/volumio-plugins/commit/1bddf32f238215bb2a884ff200db5580290ed16e 
I don't see the reason for disabling the UI on volspotconnect2. The plugin supports the WebUI buttons since version v0.5.9 of the plugin! 
@volumio  Could I request you to revet 1bddf32f238215bb2a884ff200db5580290ed16e?

volumio, 2018-04-08:
It was disabled because it was causing many crashes (can see them from the crash reports). If stability can be improven then for sure I can remove the affected commit

ashthespy, 2018-04-08:
I agree the first few versions were buggy, but it's quite stable with b165764662aa38f2a66ae61698f28746908a5778 and 4f39547c4cc6f9d2911f8650b96644b55f0b358f. Either way, I'd be happy to dig into the logs to improve the stability further!

volumio, 2018-04-09:
Thanks, please send me a mail at info at volumio dot org , so I can forward you the logs privately

ashthespy, 2018-04-23:
@volumio patch merged, should reduce those crashes once people update volspotconnect! :-)

ashthespy, 2018-05-23:
Fixed as of version `0.7.3`

closed 2018-05-23 by ashthespy