volumio/volumio-plugins #197: [No sound] with spotify apps connect to Volumio

felixlam9229, 2018-04-28:
Hi guys, Web Ui player worked all great. however when comes to Android or Windows Spotify apps, volumio Web Ui did update & play song that i picked from Spotify Apps, however there is no sound output to my speaker. below are my apps information : -

Output device : IQaudIO Plus through equalizer plugin
Volumio : 2.389
Spotify Plugin for Volumio 2 : 18-2-18
Volumio Spotify Connect2 : 20-3-18
Volumio Simple Equalizer : 5-2-18

Hope to get some helps from you guys, Spotify Apps's interface is way easier for me. Thanks

balbuze, 2018-04-28:
Is the problem for spotify or spotify connect ?
If Spotify connect : 
have you install the equalizer after volspotconnect2 ?
I found a small issue with it. I suggest you : 
- install the equalizer, enable and check it works
- install the dev version of  volspotconnect2 : https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2 and enable it

felixlam9229, 2018-05-04:
Hi Balbuze, i have no idea, had already :- 

-recreate clean & new volumio system
-install dev connect 2 - ver 0.4.0
-install spotify (official from Web Ui)
-install volumio simple equalizer

still having the same issue.. by using Web Ui control songs play is fine, once control with my Android phone, spotify apps, detected & controllable, however my speaker has no sound output.

perhaps something to do with my " Output device : IQaudIO Plus through equalizer plugin "

balbuze, 2018-05-04:
dev version of volspotconnect2 is 0.7.1
once installed, go in the plugin setting and save setting even if you don't change it.