volumio/volumio-plugins #209: volspotconnect2 not working since 2 days

megapearl, 2018-05-18:
Hi,

Just installed Volumio with spotify connect2 plugin on Raspberry Pi 3B.
'Spotify Connect Speaker' does show up in the spotify officical ios app devices list, but when you select it the square 'square volumio the audio music player' logo in the volumio gui disappears and nothing happens.
Also tried the dev version of the volspotconnect2 plugin, which is not working also.
Is this a known problem? a week ago everything did work as expected, did spotify change some things why it isn't working anymore? anybody else has this problem?

How to debug?

Best Regards,
Donald.

messismore, 2018-05-18:
Spotify Connect advertisement will break sporadically for me, but disabling and re-enabling the plugin usually solves it...

megapearl, 2018-05-18:
Spotify Connect does advertise, I can see it as source in the spotify iOS app, I can select it, but it won't stream to it, just nothing happens, reinstalled Spotify as fresh install on another pi and same problem...
Is there any log for spotify connect2 plugin where I can see what is going on?

megapearl, 2018-05-18:
Fixed for me using 3rd party plugin:

`wget https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/volspotconnect2.zip
mkdir ./volspotconnect2
miniunzip volspotconnect2.zip -d ./volspotconnect2
cd ./volspotconnect2
volumio plugin install`

Still having the errors in /var/log/volumio.log:
`2018-05-18T15:05:28.969Z - error: TypeError: Cannot read property '0' of undefined at /volumio/app/plugins/music_service/upnp_browser/index.js:83:67 at /volumio/app/plugins/music_service/upnp_browser/index.js:596:21 at Parser.<anonymous> (/volumio/node_modules/xml2js/lib/xml2js.js:489:18) at emitOne (events.js:96:13) at Parser.emit (events.js:188:7) at Object.onclosetag (/volumio/node_modules/xml2js/lib/xml2js.js:447:26) at emit (/volumio/node_modules/sax/lib/sax.js:640:35) at emitNode (/volumio/node_modules/sax/lib/sax.js:645:5) at closeTag (/volumio/node_modules/sax/lib/sax.js:905:7) at Object.write (/volumio/node_modules/sax/lib/sax.js:1452:13) at Parser.exports.Parser.Parser.parseString (/volumio/node_modules/xml2js/lib/xml2js.js:508:31) at Parser.parseString (/volumio/node_modules/xml2js/lib/xml2js.js:7:59) at /volumio/app/plugins/music_service/upnp_browser/index.js:595:24 at Request.handleRequestResponse [as _callback] (/volumio/node_modules/unirest/index.js:463:26) at Request.self.callback (/volumio/node_modules/unirest/node_modules/request/request.js:187:22) at emitTwo (events.js:106:13) `

balbuze, 2018-05-18:
errors are not related to volspotconnect2 but upnp