volumio/volumio-plugins #228: YouTube not working

albydnc, 2018-06-21:
Volumio gives 'An error occurred when browsing the folder.' when opening YouTube.

natcl, 2018-06-22:
Same issue here, search doesn't seem to work anymore :/

natcl, 2018-06-22:
Found this in the logs:
```
2018-06-22T00:40:58.170Z - info: handleBrowseUri: youtube
2018-06-22T00:40:59.383Z - info: CoreCommandRouter::executeOnPlugin: youtube , handleBrowseUri
2018-06-22T00:40:59.384Z - info: handleBrowseUri: youtube/root/likedVideos
$ usage and adjust limits in the API Console: https://console.developers.google.com/apis/api/youtube.googleapis.com/quotas?project=163050398935
Error: Daily Limit Exceeded. The quota will be reset at midnight Pacific Time (PT). You may monitor your quota usage and adjust limits in the A$
    at Request._callback (/data/plugins/music_service/youtube/node_modules/google-auth-library/lib/transporters.js:85:15)
```
Strange thing is this happened the first time I attempted a search and did not use volumio prior to this.

Felix-Franz, 2018-07-05:
I currently receive the error that the API tocken is not available:

```log
2018-07-05T07:15:54.429Z - error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
Error: Access Not Configured. YouTube Data API has not been used in project 119103001349 before or it is disabled. Enable it by visiting https://console.developers.google.com/apis/api/youtube.googleapis.com/overview?project=119103001349 then retry. If you enabled this API recently, wait a few minutes for the action to propagate to our systems and retry.
    at Request._callback (/data/plugins/music_service/youtube/node_modules/google-auth-library/lib/transporters.js:85:15)
    at Request.self.callback (/data/plugins/music_service/youtube/node_modules/request/request.js:186:22)
    at emitTwo (events.js:126:13)
    at Request.emit (events.js:214:7)
    at Request.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1163:10)
    at emitOne (events.js:116:13)
    at Request.emit (events.js:211:7)
    at IncomingMessage.<anonymous> (/data/plugins/music_service/youtube/node_modules/request/request.js:1085:12)
    at Object.onceWrapper (events.js:313:30)
    at emitNone (events.js:111:20)
    at IncomingMessage.emit (events.js:208:7)
    at endReadableNT (_stream_readable.js:1064:12)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)
```