volumio/volumio-plugins #234: TuneIn Plugin Radiostations not saveable in favorites

bamberdd, 2018-07-08:
When trying to save a radiostation via Volumio-GUI (TuneIn-plugin Updated 19.03.2018) no Dataset is added in the file "/data/favourites/favourites".  The Web-GUI confirms a successful saving of the dataset, but the Radiostation is not saved at all.

vasyugan, 2019-08-20:
Same here, but this seems to affect all plugins alike, eg podcast also also the builin internet radio source

piffio, 2019-09-03:
@volumio you can assign this one to me if you want

volumio, 2019-09-03:
@piffio sure! Assigned!

piffio, 2019-09-26:
@vasyugan and @bamberdd I just ran a test and when I save the radio stream as favourite it is saved in my `Favourite Radios` section, as you can see from the attached screenshot:

[image]

Can you provide more info on how to reproduce the issue on your side? Do you save the favourite radio from the playback screen or from within the WebRadio area?

vasyugan, 2019-11-03:
sorry, saw your response only now @piffio Will have another look in two days, when I get back home. But your screenshot doesn't look like it is the Tunein app, but the builtin internet radios app. Are you sure that you marked radio stations as favourites namely in the Tunin app?

volumio, 2019-11-04:
@vasyugan Tunein is now integrated into the internet radios facility of volumio. From there the webradios correctly adds to favourites.
We are removing therefore the tunein plugin

closed 2019-11-04 by volumio