volumio/volumio-plugins #247: Onkyo-Control plugin needs "model" for some receivers

swinokur, 2018-08-04:
Hi there,
My Integra/Onkyo receiver is several years old. It is controllable via eiscp, but clearly is a bit different than perhaps some of the other models out there.

- it doesn't support auto-discovery
- it needs the "model" string in connectionOptions

Without the "model" string, eiscp tries to discover the receiver type, and since this receiver doesn't support auto-discovery eiscp never finishes connecting and so no commands are successful.

This allows the plugin to work:
[code]
    socket.on('pushState', function (state) {
        if (self.running) {
            var connectionOptions = {
                reconnect: false,
                send_delay: 5000,
                verify_commands: false,
                model: "DTC-9.8"
            };
[/code]

Could "model" be added to the GUI? If a user is manually specifying the IP address (i.e. no auto-discovery) they probably need to manually specify the model as well.

thanks!

orderoftheflame, 2018-09-21:
Hi @swinokur, I didn't get the alert for this as it wasn't on my repo. :) I'll take a look into this when I find the chance! Soon hopefully.

orderoftheflame, 2018-09-23:
Hi @swinokur, I've raised a pull request (#262), so hopefully it'll be available shortly. Or you can go to my fork and install manually. :)