volumio/volumio-plugins #253: Autostart not .... starting anything.

houndofsound, 2018-08-21:
I've just installed 2.444 and found that perhaps the old indpendent autostart plugin (which worked) had become integrated. great! and with a start-from-last-played option now too. greater!
except it doesn't work - the playlist is there but it needs the play button pressing.

johanfohlin, 2018-08-24:
I have the same issue

nooogent, 2018-08-25:
A few things to try:

Make sure the plugin is active (on the installed plugins tab).
 
In the plugin settings, try to increase the 'Delay Before Music Starts' value to 20000, then save.

If you are using the 'start from last played' option, it looks like it will only work with local files.

johanfohlin, 2018-08-25:
It works now. Had to uninstall and reinstall.

thomastech, 2018-11-22:
> In the plugin settings, try to increase the 'Delay Before Music Starts' value to 20000, then save.

@nooogent : Just wanted to say thank you, your comment solved my problem. 

My story:
I had set the value to 10000 thinking it would shorten the music startup time. Much later, at reboot AutoPlay did not work.  Some hair pulling, followed by google, lead me to this forum. Thank goodness for community help!

- Thomas

hessercan, 2021-08-10:
How do I make this work with Pandora? I want to use this in a store. It needs to auto start pandora at startup.