volumio/volumio-plugins #261: Cannot access to Youtube plug-in Settings page

doublestat, 2018-09-09:
## Current behavior
Cannot access to Youtube plug-in Settings page. When I click `Setting` button on Youtube plug-in, just empty page shows up with this error on Chrome dev tool console:
```
TypeError: Cannot read property 'page' of null
    at app-c83c1d1c29.js:formatted:5458
    at app-c83c1d1c29.js:formatted:489
    at d.$eval (vendor-30cdd48b90.js:35)
    at d.$apply (vendor-30cdd48b90.js:35)
    at r.<anonymous> (app-c83c1d1c29.js:formatted:488)
    at r.emit (vendor-30cdd48b90.js:221)
    at r.onevent (vendor-30cdd48b90.js:221)
    at r.onpacket (vendor-30cdd48b90.js:221)
    at r.<anonymous> (vendor-30cdd48b90.js:221)
    at r.emit (vendor-30cdd48b90.js:221)
```
So I tracked up where the error happens in code inside, and found below ones. Which means client request UI config but respond with `null`, and this error happens.
```js
// app-c83c1d1c29.js (Maybe parts of Volumio?)
// ...
}, {
    key: "registerListner",
        value: function() {
            var e = this;
            this.socketService.on("pushUiConfig", function (t) {  // <-- t is null
                if (e.$log.debug("pushUiConfig", t),
                    e.pluginObj = t,
                    e.pluginObj.page.passwordProtection &&
                    e.pluginObj.page.passwordProtection.enabled) {
                        var i = "app/components/modals/modal-password.html"
// ...
```
```js
// Websocket frame logs on Chrome dev tool

4210["getUiConfig", { page: "music_service/youtube" }]
    0: "getUiConfig"
    1: { page: "music_service/youtube" }

42["pushUiConfig", null]
    0: "pushUiConfig"
    1: null
```

## Expected behavior
Some setting page shows up to config my Google account...?

## Minimal reproduction of the problem with instructions
1. Go to `Plugins`.
2. Turn on `Youtube Playback` plug-in.
3. Click `Settings` button.

## Environment
```
Raspberry Pi 3 with Volumio 2.129
Plugin: Youtube Playback
```

rovashan, 2018-11-04:
It's crashing before settings when trying to access the API because of the exceeded quota on the default account.
Use your own Youtube API key as a workaround. 

1. Create API key.
2. Edit /data/plugins/music_service/youtube/index.js.
3. Replace the value of ytapi_key with your own key.
4. Afterwards the settings page will be accessible.

Nio9345, 2019-05-29:
> It's crashing before settings when trying to access the API because of the exceeded quota on the default account.
> Use your own Youtube API key as a workaround.
>
>     1. Create API key.
>
>     2. Edit [...]

Sorry mate but this method doesn't work

adsa88, 2020-01-09:
> It's crashing before settings when trying to access the API because of the exceeded quota on the default account.
> Use your own Youtube API key as a workaround.
>
> 1. Create API key.
> 2. Edit [...]

Thx! Used this workaround before and, to this day, it's still working!(was kinda looking for this specific post from you)

In case someone reads this and wanna try, here's a quick guide how to generate your own youtube ytapi_key. It's a little outdated but there are only minor changes about menu placement.
https://elfsight.com/help/how-to-get-youtube-api-key/