volumio/volumio-plugins #286: ECONNREFUSED 127.0.0.1:6602, stack=Error: connect ECONNREFUSED 127.0.0.1:6602

cmoulliard, 2018-12-21:
Volumio version: 2.513
Spotify plugin : 1.3.5

## Error

When I add my Spotify's ID number and pasword and save it within the settings of the plugin, the following error is reported

Is it a Firewall problem as connection on port 6602 can't be established ?

```
2018-12-21T18:09:44.123Z - info: [{"prettyName":"Spotify","name":"spop","category":"music_service","version":"1.3.5","icon":"fa-spotify","enabled":true,"active":true},{"prettyName":"Youtube Playback","name":"youtube","category":"music_service","version":"0.0.9","icon":"fa-youtube","enabled":true,"active":true},{"prettyName":"Backup & Restore Data","name":"backup_restore","category":"system_controller","version":"0.7.2","icon":"fa-floppy-o","enabled":true,"active":true}]
2018-12-21T18:09:49.001Z - info: CoreCommandRouter::getUIConfigOnPlugin
2018-12-21T18:10:25.874Z - info: CALLMETHOD: music_service spop saveSpotifyAccount [object Object]
2018-12-21T18:10:25.875Z - info: CoreCommandRouter::executeOnPlugin: spop , saveSpotifyAccount
2018-12-21T18:10:26.038Z - info: SpopD Daemon Started
2018-12-21T18:10:31.039Z - info: Connecting to daemon
2018-12-21T18:10:31.050Z - info: SPOP command error:
2018-12-21T18:10:31.051Z - info:  message=connect ECONNREFUSED 127.0.0.1:6602, stack=Error: connect ECONNREFUSED 127.0.0.1:6602
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14), code=ECONNREFUSED, errno=ECONNREFUSED, syscall=connect, address=127.0.0.1, port=6602
2018-12-21T18:10:31.053Z - info: SPOP status error:
2018-12-21T18:10:31.054Z - info:  message=connect ECONNREFUSED 127.0.0.1:6602, stack=Error: connect ECONNREFUSED 127.0.0.1:6602
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14), code=ECONNREFUSED, errno=ECONNREFUSED, syscall=connect, address=127.0.0.1, port=6602
2018-12-21T18:10:31.336Z - info: Spotify access token expires at 1545419431041
2018-12-21T18:10:31.337Z - info: Spotify access token is BQBwZhGeihRGpoErWvNisu1CvbFd2bVeYlcmrpYxMqk4XxKEgJALe3aACdS-HErMQ9Yoo_el_f-xDXvN_zE
2018-12-21T18:10:31.337Z - info: Spotify credentials grant success
```

cmoulliard, 2018-12-22:
I have investigated the log of the spotify daemon and it reports that we can't use anymore `spotify` without a `premium account`

Is there a workaround ?

```
volumio@volumio:~$ systemctl status spop.service
● spop.service - Spotify Daemon
   Loaded: loaded (/lib/systemd/system/spop.service; disabled)
   Active: failed (Result: signal) since Sat 2018-12-22 12:46:14 UTC; 3s ago
  Process: 30985 ExecStart=/usr/bin/spopd -f -c /etc/spopd.conf (code=killed, signal=TRAP)
 Main PID: 30985 (code=killed, signal=TRAP)

Dec 22 12:46:14 volumio systemd[1]: Started Spotify Daemon.
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [SPTF] 12:46:14.336 I [offline_authorizer.cpp:297] Unable to login offline: no such user
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [SPTF] 12:46:14.342 I [ap:1752] Connecting to AP ap.spotify.com:4070
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [SPTF] 12:46:14.442 I [ap:1226] Connected to AP: 104.199.65.235:4070
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [SPTF] 12:46:14.651 E [ap:3953] Connection error:  409
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [WARN] Login failed: User needs a premium account
Dec 22 12:46:14 volumio volumio[30985]: 2018-12-22 12:46:14 [ERR ] Could not get the playlist container.
Dec 22 12:46:14 volumio systemd[1]: spop.service: main process exited, code=killed, status=5/TRAP
Dec 22 12:46:14 volumio systemd[1]: Unit spop.service entered failed state.
```

cmoulliard, 2018-12-26:
As the error is related to the fact that I don't use a PREMIUM account, I propose to close it. Make sense ?

volumio, 2018-12-26:
Yes, spotify works only with premium account...

closed 2018-12-26 by volumio