volumio/volumio-plugins #290: Accumulating stateMachine messages eventually slowing down everything

Kanabanarama, 2019-01-10:
I've written a plugin that starts playing music on motion detection via a PIR sensor connected to GPIO. It works as expected for a queue of tracks stored internally on my SD card. But when I use tracks of other music services as youtube or spotify, the start of the tracks will be put off more and more over time. If volumio was powered on several days, the reaction time of the player and the GUI will have become so slow that it's unbearable.

Looking at journalctl, it seems that messages are accumulating:

```
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::getState
Jan 10 14:16:35 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::setRandom true
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::pushState
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::getState
Jan 10 14:16:35 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreCommandRouter::volumioPushState
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::getQueue
Jan 10 14:16:35 badezimmer volumio[31671]: info: CorePlayQueue::getQueue
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::play index undefined
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::setConsumeUpdateService undefined
Jan 10 14:16:35 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:16:35 badezimmer volumio[31671]: info: CoreStateMachine::startPlaybackTimer
Jan 10 14:16:35 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:16:35 badezimmer volumio[31671]: info: ControllerMpd::resume
Jan 10 14:16:35 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand play
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer sudo[19285]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 10 14:16:36 badezimmer volumio[31671]: info: [ASDebug] CurState: pause PrevState: pause
Jan 10 14:16:36 badezimmer volumio[31671]: info:
Jan 10 14:16:36 badezimmer volumio[31671]: ---------------------------- MPD announces state update: player
Jan 10 14:16:36 badezimmer sudo[19285]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 10 14:16:36 badezimmer volumio[31671]: info: Youtube Status Update: player
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: Youtube Status Update: player
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: Youtube Status Update: player
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: Youtube Status Update: player
Jan 10 14:16:36 badezimmer sudo[19285]: pam_unix(sudo:session): session closed for user root
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: Youtube Status Update: player
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::getState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand status
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: sending command...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 10 14:16:36 badezimmer volumio[31671]: info: ControllerMpd::parseState
[... 328 lines cut ...]
Jan 10 14:21:43 badezimmer volumio[31671]: info: Received an update from plugin. extracting info from payload
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreStateMachine::pushState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreStateMachine::getState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreCommandRouter::volumioPushState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreStateMachine::pushState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreStateMachine::getState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CorePlayQueue::getTrack 0
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreCommandRouter::volumioPushState
Jan 10 14:21:43 badezimmer volumio[31671]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jan 10 14:21:43 badezimmer sudo[21160]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 10 14:21:43 badezimmer sudo[21160]: pam_unix(sudo:session): session closed for user root
Jan 10 14:21:43 badezimmer volumio[31671]: info: ------------------------------ 1702ms
Jan 10 14:21:43 badezimmer volumio[31671]: info: [ASDebug] CurState: play PrevState: play
Jan 10 14:21:43 badezimmer volumio[31671]: info: [ASDebug] CurState: play PrevState: play
Jan 10 14:21:43 badezimmer sudo[21166]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Jan 10 14:21:43 badezimmer sudo[21166]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 10 14:21:43 badezimmer sudo[21166]: pam_unix(sudo:session): session closed for user root
Jan 10 14:22:07 badezimmer volumio[31671]: info: [PIR_MOTION_DETECTION_AUTOPLAY]: Continue watching
Jan 10 14:22:37 badezimmer volumio[31671]: info: [PIR_MOTION_DETECTION_AUTOPLAY]: Stopping timeout triggered

```

I use the stateMachine to get statuses and to call the play / stop methods. Is there anything that has to be done differently when using other plugins music services like that or is there some magic promise fireworks going on I don't quite understand?

volumio, 2019-01-10:
From what I can see IMHO what happens is that multiple connectrions to mpd are made. Probabily in one of the newly installed plugins.
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: sending command...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...
Jan 10 14:21:42 badezimmer volumio[31671]: info: parsing response...

Kanabanarama, 2019-01-12:
I looked over it again today with only the youtube plugin active and the same thing happens:

```
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseState
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseState
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseState
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseState
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseState
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::sendMpdCommand playlistinfo
Jan 12 14:47:51 badezimmer volumio[893]: info: sending command...
Jan 12 14:47:51 badezimmer volumio[893]: info: sending command...
Jan 12 14:47:51 badezimmer volumio[893]: info: sending command...
Jan 12 14:47:51 badezimmer volumio[893]: info: sending command...
Jan 12 14:47:51 badezimmer volumio[893]: info: sending command...
Jan 12 14:47:51 badezimmer volumio[893]: info: parsing response...
Jan 12 14:47:51 badezimmer volumio[893]: info: parsing response...
Jan 12 14:47:51 badezimmer volumio[893]: info: parsing response...
Jan 12 14:47:51 badezimmer volumio[893]: info: parsing response...
Jan 12 14:47:51 badezimmer volumio[893]: info: parsing response...
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseTrackInfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseTrackInfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseTrackInfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseTrackInfo
Jan 12 14:47:51 badezimmer volumio[893]: info: ControllerMpd::parseTrackInfo
```

The mpd plugin gets assigned in the onStart function of the youtube plugin. If the plugin does not get restarted at runtime multiple times by volumio or anything, there shouldn't be multiple connections.