volumio/volumio-plugins #291: touchscreen plugin

apm1007, 2019-01-12:
Thx for this module. Bought my first official rapsberry pi touchscreen and the iqaudio pro. great job.
Here are some features i miss/ error i got. 
1. First i miss a button to change brightness. e.g. in touchscreen plugin settings?
2. got job for turning the display off automatically. Maybe this can be done buy intervall? to be set in the plugin?
3. Searching isnt possible without some keyboard. maybe to upcoming keyboard? like on smartphone?
[image]
4. default page layout is to big.
[image]
i think there are a few pixel to much on the start screen. (Grey line on the right side)
5. Cover can be a litte bigger instead of volume and play button.
;-) hope you like my suggestions in the next update! Thx.

gvolt, 2019-01-12:
Hold on a little. I am currently working on extending the plugin with configuration options for screensaver timeout, brightness setting, hiding/showing the mouse pointer and flipping the display (rotating by 180 degrees).

On screen keyboard will not be included for now. But you can get a keyboard easily by installing matchbox-window-manager and matchbox-keyboard:

`$ sudo apt-get install matchbox-window-manager matchbox-keyboard`

Then open /opt/volumiokiosk.sh with

`$ sudo nano /opt/volumiokiosk.sh`

and exchange the line

`openbox-session &`

with

`matchbox-keyboard -d &`
`matchbox-window-manager -use_titlebar no &`

Customize the /usr/share/matchbox-keyboard/keyboard.xml to the keyboard layout of your liking.

apm1007, 2019-01-24:
Thx. I will try this. Im looking fortunately to your Changes!

apm1007, 2019-02-17:
Got it. Thank you very much. Im Looking Forward to enjoy These Features.

Von: gvolt
Gesendet: Samstag, 12. Januar 2019 17:52
An: volumio/volumio-plugins
Cc: Ap Mielke; Author
Betreff: Re: [volumio/volumio-plugins] touchscreen plugin (#291)

Hold on a little. I am currently working on extending the plugin with configuration options for screensaver timeout, brightness setting, hiding/showing the mouse pointer and flipping the display (rotating by 180 degrees).
On screen keyboard will not be included for now. But you can get a keyboard easily by installing matchbox-window-manager and matchbox-keyboard:
$ sudo apt-get install matchbox-window-manager matchbox-keyboard
Then open /opt/volumiokiosk.sh with
$ sudo nano /opt/volumiokiosk.sh
and exchange the line
openbox-session &
with
matchbox-keyboard -d &
matchbox-window-manager -use_titlebar no &
Customize the /usr/share/matchbox-keyboard/keyboard.xml to the keyboard layout of your liking.