volumio/volumio-plugins #302: volspotconnect2 librespot bug

worleydl, 2019-02-22:
Recently spotify changed something on their end that breaks the ability to change tracks in librespot.  The issue has been addressed in https://github.com/librespot-org/librespot/pull/291

I've built the latest version of librespot and deployed on my pi zero and it resolves the issue, updating the builds in  https://github.com/balbuze/volumio-plugins/raw/master/plugins/music_service/volspotconnect2/arch/ should fix this.

balbuze, 2019-02-23:
Yes. But volspotconnect 2 uses a custom version of librespot in order to get metadata, hw volume control...@ashthespy seems to work on that to update it. It will be published when ready.

ashthespy, 2019-02-25:
Sorry guys, I am on vacation - will try to push changes shortly.

volumio, 2019-02-26:
Thanks @ashthespy 

PS: Is there a way we could use Volumio's REST APIs for volume changes? At least as an option to enable in librespot...

ashthespy, 2019-03-01:
@volumio sure can be done - but I don't see a way to do it with `librespot` in its current form. It could be done by running a small shell script on any volume events.

volumio, 2019-03-01:
@ashthespy that will do. Can you add that? Ideally we would need 2 events: getVolume and setVolume, I will take it from there then ;)

ashthespy, 2019-03-08:
@volumio On thinking more about this, I realised this is already implemented - [`volspotconnect2`](https://github.com/balbuze/volumio-plugins/blob/628378bac09b51444d091924d24a2b9926c67b54/plugins/music_service/volspotconnect2/index.js#L216) uses the event along with `volumioupdatevolume` to set the UI's volume.