volumio/volumio-plugins #317: Youtube plugin is not working well

BartEngelen, 2019-03-20:
Hello,
I have installed the YouTube Playback plugin and I have an issue with it.
I also installed the TuneIn Radio plugin with this plugin I can listen to radio stations very good and without any problem.
When I want to listen with the YouTube Playback plugin I can't hear the music proper, it falls away and comes back.
I can search the music with the plugin but playing it doesn't work very well.
It stucks all the time. What can I do to solve this problem ?
Kind regards,
Bart

scrallen, 2019-03-24:
I'm having major issues too with this plugin. Is returning `Error: no results`. Tried resinstall of plugin and even factory reset of volumio, only to get the same situation.

Perhaps it's yet another Youtube API change?

Using Volumio 2.565 on R-Pi3 B, Youtube Playback plugin 0.0.9.

suuuehgi, 2020-04-12:
> Is returning `Error: no results`.

Same here close to a year later. Is there anything going on with the plugin?

volumio, 2020-04-13:
see https://forum.volumio.org/youtube-for-volumio-t6608-320.html