volumio/volumio-plugins #327: YouTube playback always stops a few seconds before the end of the video/song

ThomasTheGerman, 2019-04-19:
The plugin consistently cuts off the last bit of anything I try to play

aiomaster, 2019-06-13:
I think it is exactly this problem: https://github.com/fent/node-ytdl-core/issues/402
Maybe we could workaround by downloading the file from youtube and feed the downloaded file to mpd instead of the youtube stream itself.
Or does anyone know, if there is an option for mpd to just reconnect if connection was closed?

quatmo, 2019-06-19:
> I think it is exactly this problem: [fent/node-ytdl-core#402](https://github.com/fent/node-ytdl-core/issues/402)
> Maybe we could workaround by downloading the file from youtube and feed the downloaded file to mpd instead of the youtube stream itself.
> Or does anyone know, if there is an option for mpd to just reconnect if connection was closed?

How about mpd-watchdog for reconnect? 

https://discourse.mopidy.com/t/radio-doesnt-return-after-internet-dropout-mpd-watchdog/424

http://www.runeaudio.com/forum/auto-reconnect-for-web-radio-t3278.html

https://discourse.mopidy.com/t/radio-doesnt-return-after-internet-dropout-mpd-watchdog/424/10

aiomaster, 2019-06-23:
There is some effort at mpd to integrate youtube-dl in mpd see https://github.com/MusicPlayerDaemon/MPD/pull/223
Maybe we should just wait for that and hope that this will fix these stream termination problems.
The whole volumio youtube plugin could be made a lot simpler than.
I don't know if updating mpd is a big thing in volumio, because I did not understand the build chain und wonder why volumio does not use the current version of mpd. Has anyone a hint how this works?

NuwanDhanushka, 2019-08-06:
is there any solution for this?

patrickkfkan, 2020-01-09:
We should probably close this as the issue is related to MPD and not the plugin itself. The youtube-dl node module is used for fetching metadata from the YouTube streams, but the actual playback is done by MPD. The version of MPD that Volumio comes with is outdated and does not handle the streams properly, resulting in cutoffs. You might want to try the version I have uploaded [here](https://forum.volumio.org/mpd-for-volumio-arm-armv7-and-x86-t13563.html) as an interim solution until Volumio delivers a new release based on Debian Buster (x86 version is in progress, although no ETA).

ThomasTheGerman, 2020-01-09:
ah, so the issue is debian coming with broken packages for the sake of "stability"

patrickkfkan, 2020-01-09:
No, at least not in this context. The issue is with Volumio sticking to an old version of Debian. The current version of Volumio is still based on Debian 8 Jessie, which was released 4 years ago. Debian is now at version 10 ("Buster") and the MPD package has already been updated to a recent version. Volumio, with its outdated Debian base, naturally hasn't got the updated MPD.

aiomaster, 2020-01-09:
Yeah patrickkfkan is right. The corresponding issue is this https://github.com/volumio/Build/issues/388
But maybe it could be workarounded in two ways atm:
1. compile a newer mpd for jessie (Can you give some hints, how you had done it @patrickkfkan ?)
2. Download the file with youtube-dl and feed the downloaded file to mpd and not the stream directly

I don't know which one would be more work.
At least we could wait for the buster upgrade, but waiting did not help much in the past.

patrickkfkan, 2020-01-09:
1. You can download the MPD binaries (v0.21.16) I compiled in the [link](https://forum.volumio.org/mpd-for-volumio-arm-armv7-and-x86-t13563.html) given above (instructions on activating the newer MPD also provided). Build scripts can be found in my [github](https://github.com/patrickkfkan/volumio-build-mpd).

2. This will likely work but users will notice a delay in playback of YouTube streams, particularly lengthy ones, so I wouldn't recommend it.

"waiting did not help much in the past." - I have that feeling as well....

volumio, 2020-01-09:
@patrickkfkan I tried to recompile latest mpd on Jessie, but with no avail. Would you be willing to help to produce an updated mpd package that we can integrate in current Jessie ?

patrickkfkan, 2020-01-09:
> @patrickkfkan I tried to recompile latest mpd on Jessie, but with no avail. Would you be willing to help to produce an updated mpd package that we can integrate in current Jessie ?

I made the MPD build for community testing, since I've tested only certain aspects of it and I'm not certain if is compatible with other parts of Volumio (e.g. Qobuz / Tidal...). If it is to make its way into the official builds, then it ought to have been tested well enough...so it is your call whether to proceed on this basis.

For compiling recent MPD, did you use the build scripts on my git? FYI I made my builds by running the scripts in a Debian Buster chroot. I'd be interested in knowing what issues you have in your build attempts. On the other hand, is there a reason not to use the binaries I provided?