volumio/volumio-plugins #342: lirc / nanosound_lirc failure

WarmChocolateCake, 2019-08-04:
(Not sure if this is a "volumio" or "nanosound plugin" issue.)

New / clean install of version 2.599 on Raspberry Pi 3 Model B Plus Rev 1.3

Installed nanosound plugin

Remote control not working.

Maybe a conflict of services?

```
volumio@volumio:~$ systemctl list-units | grep -i lirc

  lirc.service                loaded active exited    LSB: Starts LIRC daemon.
● nanosound_lirc.service      loaded failed failed    NanoSound LIRC Service
```

Part of journalctl:

```
Aug 04 09:58:51 volumio systemd[1]: Configuration file /lib/systemd/system/nanosound_lirc.service is marked executable. Please remove executable permission bits. Proceeding anyway.
Aug 04 09:58:51 volumio systemd-modules-load[183]: Failed to find module 'lirc_dev'
Aug 04 09:58:51 volumio systemd-modules-load[183]: Failed to find module 'lirc_rpi gpio_in_pin=17'
Aug 04 09:58:55 volumio systemd[1]: Starting NanoSound LIRC Service...
Aug 04 09:58:55 volumio systemd[1]: Started NanoSound LIRC Service.
Aug 04 09:59:00 volumio systemd[1]: nanosound_lirc.service: main process exited, code=exited, status=1/FAILURE
Aug 04 09:59:00 volumio systemd[1]: Unit nanosound_lirc.service entered failed state.
Aug 04 09:59:43 volumio sudo[1218]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/dtoverlay lirc-rpi gpio_in_pin=17
Aug 04 09:59:43 volumio sudo[1220]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop lirc.service
Aug 04 09:59:43 volumio sudo[1224]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop nanosound_lirc
Aug 04 09:59:43 volumio systemd[1]: Stopped LSB: Starts LIRC daemon..
Aug 04 09:59:43 volumio systemd[1]: Stopped NanoSound LIRC Service.
Aug 04 09:59:45 volumio volumio[1159]: info: Error enabling lirc-rpi overlay: Error: Command failed: /usr/bin/sudo /usr/bin/dtoverlay lirc-rpi gpio_in_pin=17
Aug 04 09:59:45 volumio volumio[1159]: DTOVERLAY[error]: Failed to open '/boot/overlays/lirc-rpi.dtbo'
Aug 04 09:59:45 volumio volumio[1159]: * Failed to read '/boot/overlays/lirc-rpi.dtbo'
Aug 04 09:59:46 volumio sudo[1346]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl start nanosound_lirc
Aug 04 09:59:46 volumio systemd[1]: Starting NanoSound LIRC Service...
Aug 04 09:59:46 volumio systemd[1]: Started NanoSound LIRC Service.
Aug 04 09:59:46 volumio sudo[1351]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl start lirc.service
Aug 04 09:59:46 volumio volumio[1159]: info: NanoSound lirc correctly started
Aug 04 09:59:46 volumio systemd[1]: Starting LSB: Starts LIRC daemon....
Aug 04 09:59:46 volumio lirc[1361]: Loading LIRC modules:.
Aug 04 09:59:46 volumio lirc[1361]: Unable to load LIRC kernel modules. Verify your ... failed!
Aug 04 09:59:46 volumio lirc[1361]: selected kernel modules in /etc/lirc/hardware.conf ... failed!
Aug 04 09:59:46 volumio lirc[1361]: Starting execution daemon: irexec: failed!
Aug 04 09:59:46 volumio systemd[1]: Started LSB: Starts LIRC daemon..
Aug 04 09:59:46 volumio volumio[1159]: info: lirc correctly started
Aug 04 09:59:47 volumio systemd[1]: nanosound_lirc.service: main process exited, code=exited, status=1/FAILURE
Aug 04 09:59:47 volumio systemd[1]: Unit nanosound_lirc.service entered failed state.
```

/boot/config.txt:
```
volumio@volumio:~$ cat /boot/config.txt 
dtparam=spi=on
initramfs volumio.initrd
gpu_mem=32
max_usb_current=1
dtparam=audio=on
audio_pwm_mode=2
dtparam=i2c_arm=on
disable_splash=1
hdmi_force_hotplug=1
#### Volumio i2s setting below: do not alter ####
dtoverlay=hifiberry-dacplus

#### Volumio i2s setting below: do not alter ####
dtoverlay=hifiberry-dacplus
```

lircd seems correct:
```
volumio@volumio:~$ cat /etc/lirc/lircd.conf    

# Please make this file available to others
# by sending it to <lirc@bartelmus.de>
#
# this config file was automatically generated
# using lirc-0.9.0-pre1(default) on Thu Jul  6 08:16:14 2017
#
# contributed by 
#
# brand:                       Nanomesher
# model no. of remote control: NanoSound IRRemote
# devices being controlled by this remote:
#

begin remote

  name  nanosound.conf
  bits           16
  flags SPACE_ENC|CONST_LENGTH
  eps            30
  aeps          100

  header       9190  4535
  one           543  1716
  zero          543   602
  ptrail        545
  repeat       9192  2258
  pre_data_bits   16
  pre_data       0xF7
  gap          108734
  min_repeat      1
#  suppress_repeat 1
#  uncomment to suppress unwanted repeats
  toggle_bit_mask 0x0

      begin codes
          KEY_SELECT       0x00FF
          KEY_UP           0x807F
          KEY_MUTE         0x40BF
          KEY_LEFT         0x20DF
          KEY_OK           0xA05F
          KEY_RIGHT        0x609F
          KEY_TITLE        0xE01F
          KEY_BACK         0x10EF
          KEY_DOWN         0x906F
          KEY_INFO         0x50AF
          KEY_MENU         0xD02F
          KEY_VOLUMEDOWN   0x30CF
          KEY_VOLUMEUP     0xB04F
          KEY_CHANNELDOWN  0x708F
          KEY_CHANNELUP    0xF00F
          KEY_PREVIOUS     0x08F7
          KEY_NEXT         0x8877
          KEY_REWIND       0x48B7
          KEY_FASTFORWARD  0xC837
          KEY_PLAY         0x28D7
          KEY_STOP         0xA857
          KEY_REPEAT       0x6897
          KEY_RANDOM       0xE817
      end codes

end remote
```
lircrc seems ok too:
```
volumio@volumio:~$ cat /etc/lirc/lircrc      
begin
    prog = irexec
    button = KEY_PREVIOUS
    config = volumio previous
    repeat = 0
end
begin
    prog = irexec
    button = KEY_NEXT
    config = volumio next
    repeat = 0
end
begin
    prog = irexec
    button = KEY_PLAY
    config = volumio toggle
    repeat = 0
end
begin
    prog = irexec
    button = KEY_PAUSE
    config = mpc toggle
    repeat = 0
end

begin
    prog = irexec
    button = KEY_VOLUMEUP
    config = volumio volume plus
    repeat = 4
end
begin
    prog = irexec
    button = KEY_VOLUMEDOWN
    config = volumio volume minus
    repeat = 4
end
begin
    prog = irexec
    button = KEY_FASTFORWARD
    config = volumio seek plus
    repeat = 4
end
begin
    prog = irexec
    button = KEY_REWIND
    config = volumio seek minus
    repeat = 4
end
begin
    prog = irexec
    button = KEY_MUTE
    config = mpc volume 0
    repeat = 0
end
```

Multiple errors seen:

1. service file marked as executable
2. `lirc_dev` and `lirc_rpi` modules not loaded. Some internet searches imply that these have changed to `gpio-ir` (this may be incorrect)
3. `config.txt` showing wrong dac? (twice)

Need any more information?

WarmChocolateCake, 2019-08-04:
OK, I think the issue is that the kernel has changed lirc...

so, you need to change:

1. /boot/config.txt - add the line: `dtoverlay=gpio-ir,gpio_pin=17`
2. remove the 2x `lirc` lines from /etc/modules 
3. remove the lirc module loading from nanosound.prototype.enablePIOverlay in /data/plugins/miscellanea/nanosound/index.js  (I don't think this function is needed at all)
4. change /etc/lirc/hardware.conf to use `MODULES="gpio_ir_recv"`

And then it should be working

arthurlutz, 2019-08-25:
I came accros this problem too, and am trying to get the plugin upstream to be updated https://github.com/nanomesher/Nanomesher_NanoSound/issues/5

WarmChocolateCake, 2019-09-20:
Just a quick update to say that this is still an issue on 2.609 - ie everything that I repaired from last time, is broken again.

nanomesher, 2019-09-21:
hi all,

This is actually an issue caused by kernel 4.19 upgrade since volumio 2.599.

We have done a permanent fix with a new implementation and [awaiting PR to be merged](https://github.com/volumio/volumio-plugins/pull/354)

Before it is merged and deployed to volumio repo, you can install it manually [via method 3 
](https://nanomesher.com/support/nanosound-support/)

Moverover, with this implementation you can [adapt your own remote
](https://nanomesher.com/adapting-your-own-infrared-remote-with-nanosound/)

WarmChocolateCake, 2019-09-24:
Ah, ok.
I managed to get my remote working, but I noticed that the physical buttons stopped working.
So, I'll take another look at this and see if the new instructions work.

BTW - nice addition, using our own remotes, thanks! :)

nanomesher, 2019-09-24:
For the physical buttons , make sure you choose the right DAC model . see [Configuration for your DAC](https://nanomesher.com/support/nanosound-support/)

WarmChocolateCake, 2019-10-02:
Just to confirm that using "method 3" to install the latest works well.
I'm not sure what happened previously, but the buttons are all working and the remote is working.
Thanks :)

closed 2019-10-02 by WarmChocolateCake