volumio/volumio-plugins #347: Latest Volumio, Latest Radio Paradise Plug, flooding log

jean-m-cyr, 2019-08-11:
Host: Raspberry Pi3b+
DAC: Hifiberry Plus
Volumio: 2.599
Plugin: Radio Paradise 1.10

The plugin floods /var/log/volumio.log when streaming RP FLAC.

`
2019-08-11T19:51:22.467Z - info: ERROR increasePlaybackTimer remainingTime:-11191 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.473Z - info: ERROR increasePlaybackTimer remainingTime:-11197 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.718Z - info: ERROR increasePlaybackTimer remainingTime:-11442 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.718Z - info: ERROR increasePlaybackTimer remainingTime:-11442 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.723Z - info: ERROR increasePlaybackTimer remainingTime:-11447 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.969Z - info: ERROR increasePlaybackTimer remainingTime:-11692 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.969Z - info: ERROR increasePlaybackTimer remainingTime:-11693 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:22.974Z - info: ERROR increasePlaybackTimer remainingTime:-11698 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:23.219Z - info: ERROR increasePlaybackTimer remainingTime:-11943 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:23.220Z - info: ERROR increasePlaybackTimer remainingTime:-11943 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-08-11T19:51:23.225Z - info: ERROR increasePlaybackTimer remainingTime:-11948 negative - askedForPrefetch:true - simulateStopStartDone:false
`

jean-m-cyr, 2019-09-15:
Issue persists in Volumio 2.609, playing FLAC via Radio Paradise pluggin.
```
---------------------------- MPD announces state update: player
2019-09-15T16:24:16.235Z - info: ControllerMpd::getState
2019-09-15T16:24:16.235Z - info: ControllerMpd::sendMpdCommand status
2019-09-15T16:24:16.236Z - info: sending command...
2019-09-15T16:24:16.255Z - info: 
---------------------------- MPD announces system playlist update
2019-09-15T16:24:16.255Z - info: Ignoring MPD Status Update
2019-09-15T16:24:16.256Z - info: 
---------------------------- MPD announces state update: player
2019-09-15T16:24:16.257Z - info: parsing response...
2019-09-15T16:24:16.257Z - info: ControllerMpd::getState
2019-09-15T16:24:16.258Z - info: ControllerMpd::sendMpdCommand status
2019-09-15T16:24:16.258Z - info: ControllerMpd::parseState
2019-09-15T16:24:16.258Z - info: ControllerMpd::sendMpdCommand playlistinfo
2019-09-15T16:24:16.259Z - info: sending command...
2019-09-15T16:24:16.264Z - info: sending command...
2019-09-15T16:24:16.274Z - info: ------------------------------ 11ms
2019-09-15T16:24:16.275Z - info: parsing response...
2019-09-15T16:24:16.276Z - info: parsing response...
2019-09-15T16:24:16.277Z - info: ControllerMpd::parseState
2019-09-15T16:24:16.277Z - info: ControllerMpd::sendMpdCommand playlistinfo
2019-09-15T16:24:16.278Z - info: ControllerMpd::parseTrackInfo
2019-09-15T16:24:16.279Z - info: sending command...
2019-09-15T16:24:16.280Z - info: ControllerMpd::pushState
2019-09-15T16:24:16.281Z - info: CoreCommandRouter::servicePushState
2019-09-15T16:24:16.281Z - info: CoreStateMachine::syncState
2019-09-15T16:24:16.281Z - info: ERROR Prefetch 500ms setTimeout missed >> directly calling increasePlaybackTimer
2019-09-15T16:24:16.282Z - info: ERROR this.runPlaybackTimer:true this.currentSongDuration:121 - this.currentSeek:1473 - this.prefetchDone:false - this.simulateStopStartDone:false
2019-09-15T16:24:16.283Z - info: ERROR increasePlaybackTimer remainingTime:-63 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.283Z - info: CorePlayQueue::getTrack 1
2019-09-15T16:24:16.284Z - info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current radio_paradise Received mpd
2019-09-15T16:24:16.284Z - info: ------------------------------ 51ms
2019-09-15T16:24:16.285Z - info: parsing response...
2019-09-15T16:24:16.286Z - info: ControllerMpd::parseTrackInfo
2019-09-15T16:24:16.287Z - info: ControllerMpd::pushState
2019-09-15T16:24:16.287Z - info: CoreCommandRouter::servicePushState
2019-09-15T16:24:16.287Z - info: CoreStateMachine::syncState
2019-09-15T16:24:16.288Z - info: ERROR Prefetch 500ms setTimeout missed >> directly calling increasePlaybackTimer
2019-09-15T16:24:16.288Z - info: ERROR this.runPlaybackTimer:true this.currentSongDuration:121 - this.currentSeek:184 - this.prefetchDone:false - this.simulateStopStartDone:false
2019-09-15T16:24:16.288Z - info: ERROR increasePlaybackTimer remainingTime:-6 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.289Z - info: CorePlayQueue::getTrack 1
2019-09-15T16:24:16.289Z - info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current radio_paradise Received mpd
2019-09-15T16:24:16.289Z - info: ------------------------------ 33ms
2019-09-15T16:24:16.471Z - info: ERROR increasePlaybackTimer remainingTime:-189 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.534Z - info: ERROR increasePlaybackTimer remainingTime:-252 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.538Z - info: ERROR increasePlaybackTimer remainingTime:-256 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.721Z - info: ERROR increasePlaybackTimer remainingTime:-439 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.785Z - info: ERROR increasePlaybackTimer remainingTime:-502 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.790Z - info: ERROR increasePlaybackTimer remainingTime:-507 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:16.973Z - info: ERROR increasePlaybackTimer remainingTime:-690 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.036Z - info: ERROR increasePlaybackTimer remainingTime:-754 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.040Z - info: ERROR increasePlaybackTimer remainingTime:-758 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.224Z - info: ERROR increasePlaybackTimer remainingTime:-942 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.287Z - info: ERROR increasePlaybackTimer remainingTime:-1004 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.291Z - info: ERROR increasePlaybackTimer remainingTime:-1009 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.476Z - info: ERROR increasePlaybackTimer remainingTime:-1193 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:17.538Z - info: ERROR increasePlaybackTimer remainingTime:-1256 negative - askedForPrefetch:true - simulateStopStartDone:false
[... cut ...]
2019-09-15T16:24:18.295Z - info: ERROR increasePlaybackTimer remainingTime:-2012 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:18.479Z - info: ERROR increasePlaybackTimer remainingTime:-2197 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:18.542Z - info: ERROR increasePlaybackTimer remainingTime:-2259 negative - askedForPrefetch:true - simulateStopStartDone:false
2019-09-15T16:24:18.546Z - info: ERROR increasePlaybackTimer remainingTime:-2263 
```

volumio, 2019-09-15:
Fixed with https://github.com/volumio/Volumio2/commit/972e662f152169ecedc9b70c3de715bf6bbc2e2b

closed 2019-09-15 by volumio

queengab, 2020-05-20:
Hi,

excuse me if i reopen this issue but have the same problem here:

After 20-30 minutes... radio paradise continue to shuffl music but can't hear the sound.
Host: Raspberry Pi3b+
DAC: Hifiberry Plus
Volumio: 2.7.7.3
Plugin: Radio Paradise 1.1.2

`May 20 13:27:53 volumio volumio[831]: ---------------------------- MPD announces system playlist update
May 20 13:27:53 volumio volumio[831]: info: Ignoring MPD Status Update
May 20 13:27:53 volumio volumio[831]: info:
May 20 13:27:53 volumio volumio[831]: ---------------------------- MPD announces state update: player
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::getState
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::sendMpdCommand status
May 20 13:27:53 volumio volumio[831]: info: sending command...
May 20 13:27:53 volumio volumio[831]: info:
May 20 13:27:53 volumio volumio[831]: ---------------------------- MPD announces system playlist update
May 20 13:27:53 volumio volumio[831]: info: Ignoring MPD Status Update
May 20 13:27:53 volumio volumio[831]: info:
May 20 13:27:53 volumio volumio[831]: ---------------------------- MPD announces state update: player
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::getState
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::sendMpdCommand status
May 20 13:27:53 volumio volumio[831]: info: sending command...
May 20 13:27:53 volumio volumio[831]: info: ------------------------------ 11ms
May 20 13:27:53 volumio volumio[831]: info: parsing response...
May 20 13:27:53 volumio volumio[831]: info: ------------------------------ 9ms
May 20 13:27:53 volumio volumio[831]: info: parsing response...
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::parseState
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::sendMpdCommand playlistinfo
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::parseState
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::sendMpdCommand playlistinfo
May 20 13:27:53 volumio volumio[831]: info: sending command...
May 20 13:27:53 volumio volumio[831]: info: sending command...
May 20 13:27:53 volumio volumio[831]: info: parsing response...
May 20 13:27:53 volumio volumio[831]: info: parsing response...
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::parseTrackInfo
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::parseTrackInfo
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::pushState
May 20 13:27:53 volumio volumio[831]: info: CoreCommandRouter::servicePushState
May 20 13:27:53 volumio volumio[831]: info: CoreStateMachine::syncState
May 20 13:27:53 volumio volumio[831]: info: ERROR Prefetch 500ms setTimeout missed >> directly calling increasePlaybackTimer
May 20 13:27:53 volumio volumio[831]: info: ERROR this.runPlaybackTimer:true this.currentSongDuration:234 - this.currentSeek:5115 - this.prefetchDone:false - this.simulateStopStartDone:false
May 20 13:27:53 volumio volumio[831]: info: CorePlayQueue::getTrack 4
May 20 13:27:53 volumio volumio[831]: info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current radio_paradise Received mpd
May 20 13:27:53 volumio volumio[831]: info: ControllerMpd::pushState
May 20 13:27:53 volumio volumio[831]: info: CoreCommandRouter::servicePushState
May 20 13:27:53 volumio volumio[831]: info: CoreStateMachine::syncState
May 20 13:27:53 volumio volumio[831]: info: ERROR Prefetch 500ms setTimeout missed >> directly calling increasePlaybackTimer
May 20 13:27:53 volumio volumio[831]: info: ERROR this.runPlaybackTimer:true this.currentSongDuration:234 - this.currentSeek:273 - this.prefetchDone:false - this.simulateStopStartDone:false
May 20 13:27:53 volumio volumio[831]: info: CorePlayQueue::getTrack 4
May 20 13:27:53 volumio volumio[831]: info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current radio_paradise Received mpd
May 20 13:27:53 volumio volumio[831]: info: ------------------------------ 32ms
May 20 13:27:53 volumio volumio[831]: info: ------------------------------ 28ms
May 20 13:27:58 volumio volumio[831]: info:
May 20 13:27:58 volumio volumio[831]: ---------------------------- MPD announces system playlist update
May 20 13:27:58 volumio volumio[831]: info: Ignoring MPD Status Update
May 20 13:27:58 volumio volumio[831]: info:
May 20 13:27:58 volumio volumio[831]: ---------------------------- MPD announces state update: player
May 20 13:27:58 volumio volumio[831]: info: ControllerMpd::getState
May 20 13:27:58 volumio volumio[831]: info: ControllerMpd::sendMpdCommand status
May 20 13:27:58 volumio volumio[831]: info: sending command...
May 20 13:27:58 volumio volumio[831]: info:
May 20 13:27:58 volumio volumio[831]: ---------------------------- MPD announces system playlist update
May 20 13:27:58 volumio volumio[831]: info: Ignoring MPD Status Update
May 20 13:27:58 volumio volumio[831]: info:
May 20 13:27:58 volumio volumio[831]: ---------------------------- MPD announces state update: player
May 20 13:27:58 volumio volumio[831]: info: ControllerMpd::getState
May 20 13:27:58 volumio volumio[831]: info: ControllerMpd::sendMpdCommand status
[... 1321 lines cut ...]
May 20 13:30:30 volumio volumio[831]: info: ControllerMpd::parseState
May 20 13:30:30 volumio volumio[831]: info: ControllerMpd::sendMpdCommand playlistinfo
May 20 13:30:30 volumio volumio[831]: info: sending command...
[... cut ...]
May 20 13:30:30 volumio volumio[831]: info: ERROR this.runPlaybackTimer:true this.currentSongDuration:270 - this.currentSeek:274 - this.prefetchDone:false - this.simulateStopStartDone:false
May 20 13:30:30 volumio volumio[831]: info: CorePlayQueue::getTrack 4
May 20 13:30:30 volumio volumio[831]: info: Received update from a service different from the one supposed to be playing music. Skipping notification.Current radio_paradise Received mpd
May 20 13:30:30 volumio volumio[831]: info: ------------------------------ 48ms
May 20 13:30:30 volumio volumio[831]: info: ------------------------------ 36ms
`