volumio/volumio-plugins #353: Comman failed

RpiVolumio35GPIO, 2019-09-02:
I installed the driver on a 3.5 inch GPIO Display.
And I constantly get an error.

Touch Display
Error setting screensaver timeout: Error: Command failed: /bin/bash -c "/usr/bin/xset -display :0 s off +dpms 0 0 0" No protocol specified /usr/bin/xset: unable to open display ":0"

Please tell me how to solve it or just hide error notifications.
Everything works, but when switching tracks, an error appears.
[image]

gvolt, 2019-09-02:
Did you have the touch display plugin version < 1.1.1 installed before you updated to Version 1.1.1 or newer?

If so please connect to Volumio via SSH and execute `sudo rm -r /data/volumiokiosk`. After that reinstall the plugin.

RpiVolumio35GPIO, 2019-09-02:
I did everything according to this instruction:
http://michael-peeters.blogspot.com/2018/03/

gvolt, 2019-09-02:
Ok, please try my suggestion above and don't modify the /opt/volumiokiosk.sh and /lib/systemd/system/volumio-kiosk.service after reinstalling the plugin. The plugin should let you disable the screen saver by setting the timeout value to 0. It also has an option to hide/show the mouse pointer.

Since version 1.1.1 of the plugin there is also no need to run `sudo dpkg --configure -a`  (and `sudo apt-get install -y xinit xorg openbox libexif12`).

RpiVolumio35GPIO, 2019-09-02:
did I get that right ?

install Touch display plugin
**then I do**

git clone https://github.com/goodtft/LCD-show
cd LCD-show/
sudo mkdir /etc/X11/xorg.conf.d
sudo cp ./usr/tft35a-overlay.dtb /boot/overlays/
sudo cp ./usr/tft35a-overlay.dtb /boot/overlays/tft35a.dtbo
sudo cp -rf ./usr/99-calibration.conf-35  /etc/X11/xorg.conf.d/99-calibration.conf
sudo mkdir -p /usr/share/X11/xorg.conf.d/
sudo cp -rf ./usr/99-fbturbo.conf  /usr/share/X11/xorg.conf.d/
modify the 99-calibration.conf file and add the Driver "evdev" option: 
sudo nano /etc/X11/xorg.conf.d/99-calibration.conf
the file should be:
Section "InputClass"
  Identifier "calibration"
  MatchProduct "ADS7846 Touchscreen"
  Option "Calibration" "3936 227 268 3880"
  Option "SwapAxes" "1"
  Driver "evdev"
EndSection
save the file (Control-X)
modify the /boot/config.txt
sudo nano /boot/config.txt
add the last line:
initramfs volumio.initrd
gpu_mem=32
max_usb_current=1
dtparam=audio=on
audio_pwm_mode=2
dtparam=i2c_arm=on
disable_splash=1
hdmi_force_hotplug=1
dtoverlay=tft35a
save the file (Control-X)
install missing packages:
sudo apt update
sudo apt install lightdm
sudo apt install xserver-xorg-input-evdev

**then I do**

uninstall the plugin
sudo rm -r /data/volumiokiosk
and reinstall the plugin.
reboot 

is that right ?

gvolt, 2019-09-02:
If you start from scratch (fresh install of Volumio) the steps after "then I do" should not be necessary.

If you try to solve the issue on your current system only execute the steps after "then I do".

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ sudo rm -r /data/volumiokiosk
rm: cannot remove '/data/volumiokiosk': No such file or directory

:((

gvolt, 2019-09-02:
I assume you have uninstalled the plugin now. ;)

Before reinstalling the plugin please post the output of `sudo ls -al /root` and `ls -al ~`.

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ sudo ls -al /root
[sudo] password for volumio:
total 10
drwx------ 2 root root   46 Aug  2 15:26 .
drwxrwxrwx 1 root root 4096 Sep  2 08:32 ..
-rw-r--r-- 1 root root  570 Jan 31  2010 .bashrc
-rw-r--r-- 1 root root  140 Nov 19  2007 .profile
volumio@volumio:~$ ls -al ~
total 74
drwxr-xr-x 1 volumio volumio 4096 Sep  2 08:37 .
drwxr-xr-x 1 root    root    4096 Aug  2 15:16 ..
-rw------- 1 volumio volumio   52 Sep  2 08:37 .Xauthority
-rw------- 1 volumio volumio  611 Sep  2 08:23 .bash_history
-rw-r--r-- 1 volumio volumio  220 Jan 14  2017 .bash_logout
-rw-r--r-- 1 volumio volumio 3515 Jan 14  2017 .bashrc
drwxr-xr-x 3 volumio volumio 4096 Sep  2 07:58 .cache
-rw-r--r-- 1 root    root      49 Aug  2 15:18 .cdcdrc
-rw-r--r-- 1 root    root     171 Aug  2 15:18 .cdserverrc
drwx------ 4 volumio volumio 4096 Sep  2 07:58 .config
-rw-r--r-- 1 volumio volumio   46 Sep  2 08:27 .dmrc
drwx------ 3 volumio volumio 4096 Sep  2 07:58 .pki
-rw-r--r-- 1 volumio volumio  675 Jan 14  2017 .profile
-rw------- 1 volumio volumio 3643 Sep  2 08:37 .xsession-errors
-rw------- 1 volumio volumio  888 Sep  2 08:23 .xsession-errors.old
drwxr-xr-x 6 volumio volumio 4096 Sep  2 08:04 LCD-show
volumio@volumio:~$

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ ls -al ~
total 74
drwxr-xr-x 1 volumio volumio 4096 Sep  2 08:37 .
drwxr-xr-x 1 root    root    4096 Aug  2 15:16 ..
-rw------- 1 volumio volumio   52 Sep  2 08:37 .Xauthority
-rw------- 1 volumio volumio  611 Sep  2 08:23 .bash_history
-rw-r--r-- 1 volumio volumio  220 Jan 14  2017 .bash_logout
-rw-r--r-- 1 volumio volumio 3515 Jan 14  2017 .bashrc
drwxr-xr-x 3 volumio volumio 4096 Sep  2 07:58 .cache
-rw-r--r-- 1 root    root      49 Aug  2 15:18 .cdcdrc
-rw-r--r-- 1 root    root     171 Aug  2 15:18 .cdserverrc
drwx------ 4 volumio volumio 4096 Sep  2 07:58 .config
-rw-r--r-- 1 volumio volumio   46 Sep  2 08:27 .dmrc
drwx------ 3 volumio volumio 4096 Sep  2 07:58 .pki
-rw-r--r-- 1 volumio volumio  675 Jan 14  2017 .profile
-rw------- 1 volumio volumio 3643 Sep  2 08:37 .xsession-errors
-rw------- 1 volumio volumio  888 Sep  2 08:23 .xsession-errors.old
drwxr-xr-x 6 volumio volumio 4096 Sep  2 08:04 LCD-show

gvolt, 2019-09-02:
Thanks! The permissions seem to be ok there.

Now please check if you can run xset as regular user (volumio): `/usr/bin/xset -display :0 s`

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ /usr/bin/xset -display :0 q
Invalid MIT-MAGIC-COOKIE-1 key/usr/bin/xset:  unable to open display ":0"

gvolt, 2019-09-02:
Ok, probably the xserver is not running now. Please start it with `/usr/bin/startx /etc/X11/Xsession` and try the xset command again.

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ /usr/bin/startx /etc/X11/Xsession

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux volumio 4.19.60-v7+ #1247 SMP Thu Jul 25 14:41:19 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  splash quiet plymouth.ignore-serial-consoles dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xF dwc_otg.nak_holdoff=1 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait bootdelay=5 logo.nologo vt.global_cursor_default=0 loglevel=0
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.32.6
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.2.log", Time: Mon Sep  2 09:14:08 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.

RpiVolumio35GPIO, 2019-09-02:
This display makes me cry :((

Maybe you will have time to write a detailed instruction on how to run Volumio on a 3.5 inch GPIO display ?

gvolt, 2019-09-02:
Does `/usr/bin/xset -display :0 q` work now?

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ /usr/bin/startx /etc/X11/Xsession

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux volumio 4.19.60-v7+ #1247 SMP Thu Jul 25 14:41:19 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  splash quiet plymouth.ignore-serial-consoles dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xF dwc_otg.nak_holdoff=1 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait bootdelay=5 logo.nologo vt.global_cursor_default=0 loglevel=0
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.32.6
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Mon Sep  2 09:28:30 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
^Z
[1]+  Stopped                 /usr/bin/startx /etc/X11/Xsession
volumio@volumio:~$ /usr/bin/xset -display :0
usage:  /usr/bin/xset [-display host:dpy] option ...
    To turn bell off:
        -b                b off               b 0
    To set bell volume, pitch and duration:
         b [vol [pitch [dur]]]          b on
    To disable bug compatibility mode:
        -bc
    To enable bug compatibility mode:
        bc
    To turn keyclick off:
        -c                c off               c 0
    To set keyclick volume:
         c [0-100]        c on
    To control Energy Star (DPMS) features:
        -dpms      Energy Star features off
        +dpms      Energy Star features on
         dpms [standby [suspend [off]]]
              force standby
              force suspend
              force off
              force on
              (also implicitly enables DPMS features)
              a timeout value of zero disables the mode
    To set the font path:
         fp= path[,path...]
    To restore the default font path:
         fp default
    To have the server reread font databases:
         fp rehash
    To remove elements from font path:
        -fp path[,path...]  fp- path[,path...]
    To prepend or append elements to font path:
        +fp path[,path...]  fp+ path[,path...]
    To set LED states off or on:
        -led [1-32]         led off
         led [1-32]         led on
        -led named 'name'   led off
         led named 'name'   led on
    To set mouse acceleration and threshold:
         m [acc_mult[/acc_div] [thr]]    m default
    To set pixel colors:
         p pixel_value color_name
    To turn auto-repeat off or on:
        -r [keycode]        r off
         r [keycode]        r on
         r rate [delay [rate]]
    For screen-saver control:
         s [timeout [cycle]]  s default    s on
         s blank              s noblank    s off
         s expose             s noexpose
         s activate           s reset
    For status information:  q
    To print version: -version
volumio@volumio:~$

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ /usr/bin/xset -display :0
usage:  /usr/bin/xset [-display host:dpy] option ...
    To turn bell off:
        -b                b off               b 0
    To set bell volume, pitch and duration:
         b [vol [pitch [dur]]]          b on
    To disable bug compatibility mode:
        -bc
    To enable bug compatibility mode:
        bc
    To turn keyclick off:
        -c                c off               c 0
    To set keyclick volume:
         c [0-100]        c on
    To control Energy Star (DPMS) features:
        -dpms      Energy Star features off
        +dpms      Energy Star features on
         dpms [standby [suspend [off]]]
              force standby
              force suspend
              force off
              force on
              (also implicitly enables DPMS features)
              a timeout value of zero disables the mode
    To set the font path:
         fp= path[,path...]
    To restore the default font path:
         fp default
    To have the server reread font databases:
         fp rehash
    To remove elements from font path:
        -fp path[,path...]  fp- path[,path...]
    To prepend or append elements to font path:
        +fp path[,path...]  fp+ path[,path...]
    To set LED states off or on:
        -led [1-32]         led off
         led [1-32]         led on
        -led named 'name'   led off
         led named 'name'   led on
    To set mouse acceleration and threshold:
         m [acc_mult[/acc_div] [thr]]    m default
    To set pixel colors:
         p pixel_value color_name
    To turn auto-repeat off or on:
        -r [keycode]        r off
         r [keycode]        r on
         r rate [delay [rate]]
    For screen-saver control:
         s [timeout [cycle]]  s default    s on
         s blank              s noblank    s off
         s expose             s noexpose
         s activate           s reset
    For status information:  q
    To print version: -version

gvolt, 2019-09-02:
Try the exact command `/usr/bin/xset -display :0 q` with the **q** option.

RpiVolumio35GPIO, 2019-09-02:
volumio@volumio:~$ /usr/bin/startx /etc/X11/Xsession

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux volumio 4.19.60-v7+ #1247 SMP Thu Jul 25 14:41:19 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  splash quiet plymouth.ignore-serial-consoles dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xF dwc_otg.nak_holdoff=1 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait bootdelay=5 logo.nologo vt.global_cursor_default=0 loglevel=0
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.32.6
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.2.log", Time: Mon Sep  2 09:34:55 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
^Z
[1]+  Stopped                 /usr/bin/startx /etc/X11/Xsession
volumio@volumio:~$ /usr/bin/xset -display :0 q
Invalid MIT-MAGIC-COOKIE-1 key/usr/bin/xset:  unable to open display ":0"

gkkpch, 2019-09-02:
guys, could you not start a private conversation and publish the result here instead of spamming us with mail?

gvolt, 2019-09-02:
Yes, sure. Thanks for the advice. :)

RpiVolumio35GPIO, 2019-09-02:
**I rebooted the Raspberry, via ssh then entered** /usr/bin/xset -display :0 q

volumio@volumio:~$ /usr/bin/xset -display :0 q
Keyboard Control:
  auto repeat:  on    key click percent:  0    LED mask:  00000000
  XKB indicators:
    00: Caps Lock:   off    01: Num Lock:    off    02: Scroll Lock: off
    03: Compose:     off    04: Kana:        off    05: Sleep:       off
    06: Suspend:     off    07: Mute:        off    08: Misc:        off
    09: Mail:        off    10: Charging:    off    11: Shift Lock:  off
    12: Group 2:     off    13: Mouse Keys:  off
  auto repeat delay:  660    repeat rate:  25
  auto repeating keys:  00ffffffdffffbbf
                        fadfffefffedffff
                        9fffffffffffffff
                        fff7ffffffffffff
  bell percent:  50    bell pitch:  400    bell duration:  100
Pointer Control:
  acceleration:  2/1    threshold:  4
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  600    cycle:  600
Colors:
  default colormap:  0x20    BlackPixel:  0x0    WhitePixel:  0xffff
Font Path:
  /usr/share/fonts/X11/misc,/usr/share/fonts/X11/100dpi/:unscaled,/usr/share/fonts/X11/75dpi/:unscaled,/usr/share/fonts/X11/Type1,/usr/share/fonts/X11/100dpi,/usr/share/fonts/X11/75dpi,built-ins
DPMS (Energy Star):
  Standby: 600    Suspend: 600    Off: 600
  DPMS is Enabled
  Monitor is On

**then I installed the touch display plugin and turned it on
and everything worked without errors
but after restarting the raspberry error appeared again**

RpiVolumio35GPIO, 2019-09-02:
> Yes, sure. Thanks for the advice. :)

please tell me how I can contact you ? 
email, discord, telegram ?

gvolt, 2019-09-02:
@RpiVolumio35GPIO Are you on the Volumio forum? There you can PM me (user: gvolt).

gkkpch, 2019-09-02:
or open a topic in the forum dev section :)

RpiVolumio35GPIO, 2019-09-02:
> @RpiVolumio35GPIO Are you on the Volumio forum? There you can PM me (user: gvolt).

I just registered on Volumio forum and can't write you PM

RpiVolumio35GPIO, 2019-09-02:
> guys, could you not start a private conversation and publish the result here instead of spamming us with mail?
> […](#)

Where did you see the spam here ?

The only person who agreed to help me, and now I can't talk to him because you thought it was spam. Read carefully what is spam. If you don't like this top, don't come here. This solves the real problem with running Volumio on a 3.5 inch GPIO display. If someone encounter the same problem he can read about it here. Hope for understanding, thank you friend)

gkkpch, 2019-09-02:
Please calm down, no need to get upset here.
May i carefully remind you, that this is a github issue list for developers, not a help forum!
Every message here gets sent as an email to volumio developers who registered here, I'm one of them.
We have a help forum on volumio.org (where you just registered).
There is a specific section for plugin development where you have plenty of opportunity to ask for help. I'm sure @gvolt will be happy to answer your questions there, just open new topic.

closed 2019-09-02 by gkkpch

RpiVolumio35GPIO, 2019-09-02:
Sorry, I didn't know that.
My problem is still unresolved.