volumio/volumio-plugins #381: webradio playback status problem (conflicts with volspotconnect2)

dmnc-net, 2019-11-19:
**Steps to reproduce:**
1. with Spotify Connect (using @balbuze `volspotconnect2`) start playing a song
2. in Volumio-UI overtake the playback by playing some `webradio` stream
3. watching via CLI `volumio status` ... results in Symptoms section
4. at this moment other plugins like thinks there is no playback and trying to play its content but webradio is still playing

**Symptoms:**
CLI command `volumio status`:
`"status"` became `"play"` for a quick moment and then turns to `"stop"` but music is playing and `"seek"` is changing. Also icon in the UI shows ▶ like if there is no playback.

**Background:**
If Volumio is used by multiple users, some of them are using Volumio UI for webradio, other just Spotify Connect thru plugin, this could be a issue.

I'm not sure if this is volspotconnect issue because I'm not using any other plugin (youtube and Volumio's Spotify doesn't work at all).

Screencast: https://youtu.be/GSf4fcc1jCo

ashthespy, 2019-11-20:
@dmnc-net What version of Volumio are you on? This shouldn't happen, and the sink should be released when a different source requests playback.

dmnc-net, 2020-01-31:
Oh, silly me, I forgot to mention the version which was 2.668.

Now, with v.2.699 and volspotconnect2 v.0.9.9 problem still remains

ashthespy, 2020-01-31:
@dmnc-net Could you share your logs when this happens?

unvirtual, 2021-04-07:
I think the issue I observe is strongly related to this one.

**Steps to reproduce:**

1. with Spotify Connect (using volspotconnect2) start playing a song (from Android phone)
2. in Volumio-UI overtake the playback by playing some webradio stream. Webradio plays correctly.
3. Spotify client on Android phone pauses and still appears to be connected to volumio (green icon). Not sure if correct.
4. Hitting play on Spotify client again doesn't stop webradio and exception below is thrown.

**Symptoms**
Switching between spotify connect and webradio back and forth doesn't work reliably.

Volumio v.2.873
volspotconnect2 v1.0.5

```
Starting Live Log...
[SpotifyConnect]  62.0004577706569
[SpotifyConnect] Volume: Spotify:62.0004577706569 Volumio: 62
info: CoreStateMachine::pushState
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
info: CoreCommandRouter::executeOnPlugin: system , getSystemVersion
info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
info: CALLMETHOD: system_controller system enableLiveLog true
info: CoreCommandRouter::executeOnPlugin: system , enableLiveLog
info: Launching a new LiveLog session
[Vollibrespot] : Event: Pause { track_id: SpotifyId { id: 102876374306587101830094222322942557107, audio_type: Track }, position_ms: 114038 }
[SpotifyConnect]  pause
[SpotifyConnect]  Fracture
[SpotifyConnect] Pushing metadata
info: CoreCommandRouter::volumioGetState
[SpotifyConnect] Currently active: volspotconnect2
[SpotifyConnect] Pushing new state :: true
info: CoreCommandRouter::servicePushState
info: CoreStateMachine::pushState
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
[Vollibrespot] : Event: SinkInactive
[Vollibrespot] : Event: PlaybackStopped { track_id: SpotifyId { id: 102876374306587101830094222322942557107, audio_type: Track } }
[SpotifyConnect]  Sink released
info: CoreCommandRouter::servicePushState
info: CoreStateMachine::pushState
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
[SpotifyConnect]  Device palyback is inactive
[SpotifyConnect] Device is paused
[Vollibrespot] : Event: Play { track_id: SpotifyId { id: 102876374306587101830094222322942557107, audio_type: Track }, position_ms: 114495 }
[SpotifyConnect]  play
[Vollibrespot] : Event: SinkActive
[SpotifyConnect]  Fracture
[SpotifyConnect] Pushing metadata
info: CoreCommandRouter::volumioGetState
[SpotifyConnect] Currently active: volspotconnect2
[SpotifyConnect] Pushing new state :: true
info: CoreCommandRouter::servicePushState
info: CoreStateMachine::pushState
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
[SpotifyConnect]  Sink acquired
info: CoreCommandRouter::volumioGetState
[SpotifyConnect] Currently active: volspotconnect2
[SpotifyConnect] Pushing new state :: true
info: CoreCommandRouter::servicePushState
info: CoreStateMachine::pushState
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
info: CoreCommandRouter::volumioReplaceandPlayItems
info: CoreStateMachine::ClearQueue
info: CoreStateMachine::stop
[... 246 lines cut ...]
at src/libstd/panicking.rs:198
9:   0x9d5640 - std::panicking::default_hook::h46ab82039cbc65eb
at src/libstd/panicking.rs:217
10:   0x9d6054 - std::panicking::rust_panic_with_hook::h7326c48419bc7c33
at src/libstd/panicking.rs:526
11:   0x9d5c54 - rust_begin_unwind
at src/libstd/panicking.rs:437
12:   0x9f1594 - core::panicking::panic_fmt::ha292e19d5ae716ed
at src/libcore/panicking.rs:85
13:   0x9f14f0 - core::panicking::panic::heb457f4fa750842b
at src/libcore/panicking.rs:50
14:   0x5d0ee8 - ::write::h5da62cd8da1289c3
15:   0x5aa3c4 - ::poll::h2bafd5dc8d40bf3f
16:   0x4baeec - futures::task_impl::std::set::h2116cb0e9a5165fe
17:   0x4afc6c - futures::task_impl::Spawn::poll_future_notify::h20a5cc82ac41f2fa
18:   0x4c96a0 - futures::future::Future::wait::ha8819dd7a48bc248
19:   0x4b14b4 - std::sys_common::backtrace::__rust_begin_short_backtrace::habfd768f2a80dc30
20:   0x4d05a8 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h0dcfa92cb8b4ff47
21:   0x9d946c -  as core::ops::function::FnOnce>::call_once::hb2bcc0b96e754d10
at /rustc/04488afe34512aa4c33566eb16d8c912a3ae04f9/src/liballoc/boxed.rs:1081
22:   0x9d946c -  as core::ops::function::FnOnce>::call_once::h2dd73f7c66408260
at /rustc/04488afe34512aa4c33566eb16d8c912a3ae04f9/src/liballoc/boxed.rs:1081
23:   0x9d946c - std::sys::unix::thread::Thread::new::thread_start::h6fc0046e32ed87dc
at src/libstd/sys/unix/thread.rs:87
```