volumio/volumio-plugins #385: Touch screen not correctly detected

bertmelis, 2019-12-03:
I have a RPI3 with the official touch screen. On rotating the screen via the web ui, the display rotates, but the touch does not.

Symptom/effect is here:
https://github.com/volumio/volumio-plugins/blob/4a6bf3bc5428ce78de4fae28f84c048d7e490f2d/plugins/miscellanea/touch_display/index.js#L975-L980

```
$ /bin/grep "^rpi_ft5406\\>" /proc/modules
rpi_ft5406 16384 0 - Live 0x00000000
```

gvolt, 2019-12-03:
I will look into soon. Are you a member of the Volumio forum, so I could reach you there by PM if necessary?

bertmelis, 2019-12-03:
Not on the forum (yet).

bertmelis, 2019-12-04:
my account name is BertM, but I cannot send PMs yet according forum's policy: "a need to practicce more".