volumio/volumio-plugins #386: Touch screen, wake-up only on first touch

bertmelis, 2019-12-09:
After the  RPI screen goes dark, the first touch not only wakes the screen but also executes the action attached to the region where pushed. As the screen is dark, you cannot see what will happen.

This is annoying behaviour.

Is it possible to have the first touch only wake up the screen without any command?

danowar2k, 2020-04-26:
This is my problem as well. My 6-year-old daughter uses the touchscreen and doesn't always remind herself to touch a certain spot on the screen where nothing would happen. It would be really nice if there was an option that could prevent the action when the screen is deactivated.

Is this even technically possible? If yes, I could try a pull request...

danowar2k, 2020-04-26:
I've searched for an event like beforePushState or something to have to something to latch onto to prevent further state changes when the display. No luck so far...

A link to someone trying to bypass the same problem:
https://unix.stackexchange.com/questions/265557/ignoring-first-click-event-on-dpms-or-screen-wakeup

danowar2k, 2020-04-26:
So, maybe it would be a solution to not just turn the screensaver on after 120 seconds, but instead switch to a different console after 120 seconds and there turn the screensaver on. Then when a touch is registered, switch back to the volumio GUI and reset the timeout.

What do the maintainers think? Would that be reasonable?

danowar2k, 2020-04-26:
I'll just collect more info here for now:

https://github.com/volumio/Volumio2-UI/blob/master/src/app/services/player.service.js
This file from the UI repo or rather the function `registerListner` seems relevant, at least if there would be a way to block event propagation somehow.

gvolt, 2020-04-26:
The plugins originates from Volumio. In last months I reworked most of the node js part (index.js).

> EDIT: It looks like this file here contains the code with the "problem":
> 
> https://github.com/volumio/volumio-plugins/blob/master/plugins/miscellanea/touch_display/index.js
> 
> Or more specifically, this code here:
> 
> ...
> 
> I don't know when "pushState" is emitted, but I guess it's when a button is pushed in the UI (play/pause/change to lists/settings/etc.)
> 
> The code reads like this: When some action (pushState) arrives, reset the screensaver (with some configured conditions).
> What I don't know is whether one can stop propagating the pushState event from here.

"pushState" is emitted not only on touching the local (touch)screen but also on commands send by IR remotes or on actions that happen in a browser running on another device (mobile, tablet, PC...).

AFAIK from the events in the Volumio system it is not possible to differentiate from where (from which input device) a change of the playing state originates. So IMHO changing the code above would need another "source" from which the origin of an input event can be determined. I guess this would involve detecting what input device the touchscreen is assigned to and then reading input events from this device.

I have been experimenting with this approach some time ago, but did not finish because lack of time. As soon as time allows, I will have a look at it again.

danowar2k, 2020-04-26:
> I have been experimenting with this approach some time ago, but did not finish because lack of 
> time. As soon as time allows, I will have a look at it again.

Thank you for commenting! 

Hm, if pushState is emitted by many devices, is there maybe a way to go between touching the touchscreen and emitting pushState? Maybe that would involve the X display screensaver and that would be too low level? Hmm....