volumio/volumio-plugins #387: YouTube code undefined

gvijayakumar-16, 2019-12-10:
I recently installed Volumio and installed youtube plugin as well. But I see an empty box where code should be. Is this a bug or am I missing something?

[image]

gvijayakumar-16, 2019-12-10:
Adding the below workaround:

https://forum.volumio.org/youtube-for-volumio-t6608-160.html
(See post by by patrickkfkan » Sun Nov 10 2019 09:43)

noeroti3, 2019-12-12:
I am also.
[image]

severindupouy, 2019-12-14:
Same for me. Can't use the plugin, there is no code.
[image]

Pit-Storm, 2019-12-23:
The above mentioned workaround works fine.

When will this be in official release?

PurelyNicole, 2020-01-06:
Confirming the [workaround](https://forum.volumio.org/youtube-for-volumio-t6608-160.html#p68753) worked for me as well.

aiomaster, 2020-01-08:
I think the modifications of @patrickkfkan make sense and at least work for me. Would it be possible to integrate [his work](https://github.com/patrickkfkan/volumio-plugins) back into upstream or is there another solution planned?

patrickkfkan, 2020-01-09:
It appears they are going to do authentication in the cloud like the Spotify plugin. We won't know how it would turn out in the end, though. Progress has been slow...

aiomaster, 2020-01-09:
But is this a solution to the "reaching request limit for this api account"-problem?
You solved it by creating the possibility to setup your own app with api key, that is unlikely to reach the limits, right?

patrickkfkan, 2020-01-09:
By setting up your own app with API key, you are much less likely to run into the "request limit" problem. There is still a limit imposed by Google but since you are the only one using the quota allocated to your own API key, this limit is much less likely to be reached under normal usage.

aiomaster, 2020-01-09:
The latest official commit I've seen so far was https://github.com/volumio/volumio-plugins/commit/b98c4099492a233e180eb47fddc2ca10eb4288fc from @crisp00 
Maybe he is able to tell more if and how the request limit issue should be addressed by this.

I think it would be cool to have your improvements as alternative to that. I've seen you implemented some kind of cache also. So it would be cool to have that stuff upstream. Do you want to rebase your work to the current master and make a pull request, @patrickkfkan ?

patrickkfkan, 2020-01-09:
I think at this stage we should probably wait until they release a new official version, then based on that I can make a PR. Apart from caching, I also implemented pagination, and I think metadata fetching can be optimized a bit more (but too lazy for that now) :).

elephantaplastee, 2020-02-05:
Hi Guys. Bill here, in N.Zealand. I recently bought a D.A.C for my Pi 3b+, and installed Volumio. I subscribed to Spotify, and also to YouTube. Please excuse me in advance, I'm very new to all this. I managed to install the YouTube plugin, and I'm really amazed, because I had it all up and working, but then unfortunately lost it. I installed the Spotify plugin with relative ease, and that's running fine at the moment, but the YouTube plugin is really doing my head in. ! ;^) I've tried lots of different ways, and have pretty much given up. I'm obviously doing something wrong. The plugin is connecting, and showing still images of various YouTube sites, but that's all. the Google Api/OAuth thing for me, is a complete nightmare, having to negotiate round those pages, with protocols, and regulations. Anyone could be forgiven for thinking this is top secret stuff. ! I'm just trying to get my Pi going. ! ;^) Sorry.. I musn't rant. If anyone can offer any help, I'd be extremely grateful. I can supply any information that might be required. Volumio's great, it just seems to lack the community support, but I'm sure it'll happen. thanks guys.

elephantaplastee, 2020-02-10:
Hello guys again. Has anyone any more news about a possible update fo the youtube plugin. My Spotify is fine, bit the youtube plugin just isn't happening. I've tried burning different sd cards, without any success. I anyone else experiencing problems. ? Thanks.

patrickkfkan, 2020-02-11:
Hi @elephantaplastee ,

Is it possible for you to drop a message on the Volumio forums under the [YouTube plugin thread](https://forum.volumio.org/youtube-for-volumio-t6608.html)? Please also mention what error messages, if any, you're getting. 

-Patrick

elephantaplastee, 2020-02-12:
Thanks very much for replying Patrick. That's very kind of you. I'm
terribly sorry for my late reply.
I just checked my email, and got quite a pleasant surprise when I saw
yours...
I'll get back to you in a couple of hours with some details. Pleas excuse
me, I'm really slow.
Thanks again.

elephantaplastee, 2020-02-13:
Hi Patrick.
       Thanks again m8. Sorry, but I'm having trouble logging into the forum, wrong password, or ?
I followed your instructions re. installing the youtube plugin in Volumio, and now have audio on some clips, but definitely no video ? I'm just a bit confused, because as I said in my last email, everything was working, and it was great, but then, I've obviously changed something, and now have no video. ? It's just really daunting. There are, to me, so many variables involved, I don
't know where to begin. ?! Could be something with Google permissions ? I just don
't know. I'll have to get my password sorted out in the Forum, and get some help. Thank you so much for your assistance m8. I can see you are helping others out in the forum. What a job. ! You must have a lot of patience... Thanks again. So sorry late getting back to you.

elephantaplastee, 2020-02-13:
Hi. I'm getting "Object Error" Messages now. ?
Never had them before. ?

elephantaplastee, 2020-02-13:
Again, yea, hi. "Error" [Object Object]
This is a new one Patrick. I have no idea. 
I've uninstalled, re-installed, formatted etc. etc till I'm blue in the face.
After I've done all that.. It's now coming up with that error. ?
I'll try another sd card. ?

elephantaplastee, 2020-02-13:
Nope.. Same thing..
ERROR [object Object]

elephantaplastee, 2020-02-13:
I've somehow blundered my way into getting a copy of my /var/log/volumio.log. Not quite sure where to post that. ? Here ?..
There's errors here..
2020-02-13T05:55:26.919Z - info: CoreCommandRouter::executeOnPlugin: my_volumio , setMyVolumioToken
2020-02-13T05:55:36.859Z - error: error
2020-02-13T05:55:36.882Z - error: error
2020-02-13T05:56:26.853Z - error: error
2020-02-13T05:56:26.876Z - error: error
2020-02-13T05:57:16.856Z - error: error
2020-02-13T05:57:16.879Z - error: error
Sorry Patrick, I really have no clue here. Hope this helps m8. Thanks... All the best..

[Z.txt](https://github.com/volumio/volumio-plugins/files/4196831/Z.txt)

patrickkfkan, 2020-02-13:
Hi @elephantaplastee ,

The log file you attached does not seem relevant. There were no references to the Youtube plugin at all (on the other hand, I do see something related to Spotify). 

You said you got audio on some YouTube clips, so it would seem the API keys were setup correctly. Have you been able to perform a search with the YouTube plugin as well? Were there any clips that refused to play?

There is definitely no video - Volumio is purely audio and the YouTube plugin merely plays the audio stream of clips you select.

-Patrick

elephantaplastee, 2020-02-13:
Ahh... So sorry about all that confusion. I've been wasting your time.
I was running Kodi at some point as well, and probably got confused about
the video thing.
Otherwise Patrick, Volumio is doing what it's supposed to be doing.
I have had a few problems with some clips stopping short.. i.e. not
playing the full duration of the track ? but I'm not sure if that's a
Volumio problem,
or my modem, or router. ? Thanks so much for taking the time Patrick.
You're doing a great service m8.
In this day and age people don't bother to do what you do. !
Sorry about the log thing. I honestly haven't a clue. !
If there's anything I can send you to help fine tune my Volumio... please
let me know.
Thanks again m8. Cheers.
                                          Bill
p,s, Sorry, yes, I was able to do a search ok with the youtube plugin.
As far as "refused to play" ? some of them just didn't... play. ?

patrickkfkan, 2020-02-15:
@elephantaplastee, could you confirm you are using the [latest version (20191224)](https://forum.volumio.org/youtube-for-volumio-t6608-230.html#p70392) I have made available in the forums? Clips might randomly fail to play with older versions.

For the problem with clips stopping short, you can follow the Github issue here: [https://github.com/volumio/volumio-plugins/issues/327](url) , and try out a later version of MPD I have made available [here](https://forum.volumio.org/mpd-for-volumio-arm-armv7-and-x86-t13563.html). Make sure you download the binary that corresponds to your device. 

-Patrick

elephantaplastee, 2020-02-15:
Thanks for that Patrick... Will do now... Cheers.

elephantaplastee, 2020-02-15:
and sorry patrick... i can't get through to http://54.86.144.136:8888/
anymore. ?

patrickkfkan, 2020-02-15:
> and sorry patrick... i can't get through to http://54.86.144.136:8888/ anymore. ?

This could be related to the Spotify plug-in which to my knowledge accesses this IP for authentication stuff.

elephantaplastee, 2020-02-16:
sure m8... it's doing my head in here... i can't.. it won't allow me to
"authorize" and says something about specifying scopes ?

elephantaplastee, 2020-02-16:
I've just about given up. They, whoever "they" are... have made this thing
so bloody convoluted. !?

elephantaplastee, 2020-02-16:
I just can't simply put my login/password in anymore... it's just nuts m8.

elephantaplastee, 2020-02-16:
You must specify a valid list of scopes...
That's what it's telling me.

elephantaplastee, 2020-02-17:
Patrick. Hi. I don't suppose there's any news on the
http://54.86.144.136:8888, thing. ?
I've followed your instructions from the forum, and everything's working as
it should,
apart from Spotify. I just can't fathom how on earth to get a "Refresh
Token".
It's just me. There's far too much information in there for just a simple
password.
However you had it set up before, was extremely simple.
I'll just keep reading m8. Thanks Patrick.

patrickkfkan, 2020-02-18:
@elephantaplastee , I have no knowledge of how the Spotify plugin works (given that I don't have a Spotify account myself), so I'm afraid I won't be able to help you on this issue. 

I suggest you create a new issue on Github instead of posting under the current one since it's not YouTube-related, or head off to the Volumio forums and seek assistance there.

volumio, 2020-03-09:
Current Youtube plugin has been removed since we need to request an additional QUOTA from youtube. Nothing we can do at the moment more of that.