volumio/volumio-plugins #398: Snapcast plugin not showing up in plugins list

faultfactory, 2020-01-12:
I can't find any means of refreshing / filtering / pulling the plugins list. 
I have 1 RPI 2 + hifiberry amp i intend to setup as a snapcast client. 
I run a VM with the volumio x86 build that i intend to run as the server for multi-room playback once i get this up and running. 

The snapcast plugin shows in the list for the RPI2. Is there any reason it would not show up on the VM? I currently have no audio device, simulated or pass through as i intended it to just be a server.

faultfactory, 2020-01-14:
I believe from reading other posts this is intentional.

Is there any remedy to this I can contribute to?

balbuze, 2020-01-15:
It is because the plugin is not 'ready' to work on x86 arch. Maybe you can contact the author...