volumio/volumio-plugins #409: Touch Screen calibration

SuperManth, 2020-01-29:
This is a great plugin. That works fine with my with my RPi3b+ and Waveshare 5inch HDMI LCD(B). Only the calibration for the pen is bad. I can not get to the corners. So it will be good idea to have a calibration function somewhere.
I have seen there is already a request for this in the forum but no one has replied on that since 2018.

I guess the idea of a touchscreen is not to use an extra mouse or keyboard. But now I need this to shutdown the system or to configure something.

Mario

gvolt, 2020-01-29:
Nice to get positive feedback on the plugin :)

Calibration of the touch function of displays is beyond the scope of the plugin IMHO. But according to the [Waveshare wiki](https://www.waveshare.com/wiki/5inch_HDMI_LCD_(B)) you should be able to calibrate your display with the following steps:

Install xserver-xorg-input-evdev and xinput-calibrator:
`sudo apt-get install xserver-xorg-input-evdev`
`sudo apt-get install -y xinput-calibrator`

Then run
`sudo DISPLAY=:0.0 xinput_calibrator`

After following the calibration procedure the generated calibration data will be displayed in the terminal. Copy or write down these values and open /etc/X11/xorg.conf.d/99-calibration.conf:
`sudo nano /etc/X11/xorg.conf.d/99-calibration.conf`

Look for the line beginning with `Option	"Calibration"` and subsitute the calibration data with the new values you copied before.

Press the Ctrl+X then Y to save 99-calibration.conf.

Then reboot.

gkkpch, 2020-01-29:
> Calibration of the touch function of displays is beyond the scope of the plugin IMHO. But according to the Waveshare wiki you should be able to calibrate your display with the following steps:

second that, calibration and tools used vary from screen to screen, this is very difficult to create a generic solution for. Would otherwise call for screen-specific plugin extensions or a specific calibration plugin.