volumio/volumio-plugins #420: Spotify Connect Plugin does not respect Max Volume Level

Rick-T, 2020-03-03:
Hi,

2 days ago I have installed Volumio 2 on a Raspberry Pi 3B+. After setup I installed the Spotify Connect Plugin. Otherwise I'm on a fresh install.

I have connected the Khadas Tone Board as a USB DAC which feeds directly into a power amp. Because I am not connecting through a pre-amp I have to use the hardware volume control of the DAC in volumio. I have also set a max volume level for safety.

Overall I have set the following playback options:

Audio Output:
> Output Device: Khadas Tone Control
> I2S DAC: Off

Volume Options:
> Mixer Type: Hardware
> Mixer Control Name: Khadas Tone Control
> Default Startup Volume: 10
> Max Volume Level: 20

In Volumio the volume control works and can not go above the max volume level.
When I am streaming music through Spotify Connect (using my Smartphone as "remote") I can use the volume buttons on my phone to control Spotify's playback volume. But when I do that the Max Volume Level of Volumio is not respected. I can go up to 100%.
Afterwards changing the volume in Volumio limits the volume again to the max level.

Being able to circumvent the safety of the max volume this way is a potential hazard to speakers and (even worse) human hearing. Especially since the Spotify app seems like it remembers the volume it was set to. So with a single accidental push of the volume button you can jump to 100% volume and blow your speakers or eardrums.

Is there any configuration options that I am overlooking or is this a problem with the plugin? If so, is there anything you can do about it?

ashthespy, 2020-03-06:
Good point, I will have to implement this in the [`vollibrespot`](https://github.com/ashthespy/Vollibrespot).

volumio, 2020-03-06:
@ashthespy I would suggest the volumio integration approach, this way behaviours will be consistent will all volumio mechanichs.
You can get and set the volume via rest api calls. This will give us the best integration possible (also for hardware which does not rely on alsa).
What do you think?

ashthespy, 2020-03-06:
IIRC, I looking into this, and implemented the socket plumbing to communicate with the plugin interface, but it was proving to be quite a laggy experience in an already large chain:
Spotify client -> Spotify Sever -> Connect client 
   -> Volumio Plugin -> Volumio Volume Controller -> amxier 

That being said, I didn't bother optimising anything, as I don't have a DAC with hardware volume control to test.. 

Do you have an example of hardware that doesn't rely on ALSA? I am guessing it's some custom rolled i2c based controller?

volumio, 2020-03-06:
IMHO  by using the REST api you remove at least 2 steps in the chain and it generally works pretty fast. Can we have a quick POC ?

ashthespy, 2020-03-06:
TBH, implementing a REST into the daemon is a can of worms I don't particularly feel like approaching right now - I already expose the volume events (https://github.com/volumio/volumio-plugins/issues/302#issuecomment-470987195) via a socket, but that still uses ALSA to fetch the volume. 

PoC as discussed: [vollibrespot-armv7l.zip](https://github.com/volumio/volumio-plugins/files/4299418/vollibrespot-armv7l.zip)

drewbroadley, 2022-01-10:
Any update on this?

volumio, 2022-01-14:
@ashthespy what about invoking volumio volume command?

This will make sure we keep consistent options