volumio/volumio-plugins #423: Rotary encoder plugin not storing configuration after reset

dubaleeiro, 2020-03-08:
Hello,

for some reason I can't understand, the rotary encoder configuration (pins, detent action, etc) is "lost" after rebooting the Pi and only gets back to work when the configuration is saved in the UI.

I have noticed the file '/data/configuration/user_interface/rotaryencoder/config.json' changes accordingly, but the file '/data/plugins/user_interface/rotaryencoder/config.json' don't. Well, I tried to change this one manually and it didn't work either.

Can anyone help me on that? Thank you!

Raspberry Pi Zero W V1.1
Innomaker HiFi AMP Hat
System Version: 2.714
Plugin Rotary Encoder 1.2.2 (maybe obvious information after the one above...)
Encoder #1: CLK=24; DT=23;detent action=1 (volume control)
Encoder #2: CLK=8; DT=25;detent action=2 (prev/next)

balbuze, 2020-03-08:
hello,
 '/data/plugins/user_interface/rotaryencoder/config.json'  is the default config file when you install the plugin. You are not supposed to change it.
 '/data/configuration/user_interface/rotaryencoder/config.json' is the file where config is saved when you save in the config page. So if your values are save here, everything is ok!

dubaleeiro, 2020-03-08:
Hi,
values are correctly saved in the '/data/configuration/user_interface/rotaryencoder/config.json' file. The problem is that after the Pi is reset the only way to make the rotary encoder to work is to ckick 'save' in the config page again. Any possible reasons for that?

This is the LOG file, if it helps...
Thanks!

`volumio@volumio:~$ sudo journalctl -f
-- Logs begin at Sun 2020-03-08 15:18:14 UTC. --
Mar 08 15:19:58 volumio systemd[915]: Startup finished in 192ms.
Mar 08 15:19:58 volumio systemd[1]: Started User Manager for UID 1000.
Mar 08 15:19:58 volumio sshd[899]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Mar 08 15:19:58 volumio sshd[899]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Mar 08 15:19:58 volumio volumio[867]: info: Loading plugin "upnp_browser"...
Mar 08 15:20:00 volumio volumio[867]: info: Loading plugin "networkfs"...
Mar 08 15:20:01 volumio volumio[867]: info: CoreCommandRouter::executeOnPlugin: system , registerCallback
Mar 08 15:20:01 volumio volumio[867]: info: Loading plugin "alarm-clock"...
Mar 08 15:20:01 volumio sudo[930]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Mar 08 15:20:01 volumio sudo[930]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Mar 08 15:20:02 volumio volumio[867]: info: Loading plugin "airplay_emulation"...
Mar 08 15:20:02 volumio volumio[867]: info: Starting Shairport Sync
Mar 08 15:20:02 volumio volumio[867]: info: Loading plugin "last_100"...
Mar 08 15:20:02 volumio volumio[867]: info: Loading plugin "webradio"...
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "i2s_dacs"...
Mar 08 15:20:03 volumio volumio[867]: info: CoreCommandRouter::executeOnPlugin: system , getConfigParam
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "volumiodiscovery"...
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Mar 08 15:20:03 volumio node[867]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Mar 08 15:20:03 volumio node[867]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 08 15:20:03 volumio node[867]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Mar 08 15:20:03 volumio node[867]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Mar 08 15:20:03 volumio node[867]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 08 15:20:03 volumio node[867]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 08 15:20:03 volumio volumio[867]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Mar 08 15:20:03 volumio volumio[867]: info: Applying required configuration parameters for plugin volumiodiscovery
Mar 08 15:20:03 volumio volumio[867]: Discovery: StartAdv! undefined
Mar 08 15:20:03 volumio volumio[867]: Discovery: Started advertising... Volumio - undefined
Mar 08 15:20:03 volumio volumio[867]: info: CoreCommandRouter::executeOnPlugin: system , registerCallback
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "outputs"...
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "albumart"...
Mar 08 15:20:03 volumio volumio[867]: info: Plugin example_plugin is not enabled
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "inputs"...
Mar 08 15:20:03 volumio volumio[867]: info: Loading plugin "updater_comm"...
Mar 08 15:20:05 volumio volumio[867]: info: Plugin mpdemulation is not enabled
Mar 08 15:20:05 volumio volumio[867]: info: Loading plugin "rest_api"...
Mar 08 15:20:05 volumio volumio[867]: info: Loading plugin "websocket"...
Mar 08 15:20:05 volumio volumio[867]: info: Loading plugin "spop"...
Mar 08 15:20:06 volumio volumio[867]: Forking 1 albumart workers
Mar 08 15:20:09 volumio volumio-remote-updater[469]: [2020-03-08 15:20:09] [connect] Successful connection
Mar 08 15:20:12 volumio volumio[867]: info: Loading plugin "rotaryencoder"...
Mar 08 15:20:12 volumio volumio[867]: Starting albumart workers
Mar 08 15:20:15 volumio volumio[867]: info: Applying required configuration parameters for plugin rotaryencoder
Mar 08 15:20:16 volumio volumio[867]: info: ___________ START PLUGINS ___________
Mar 08 15:20:16 volumio volumio[867]: info: CoreCommandRouter::executeOnPlugin: mpd , getConfigParam
Mar 08 15:20:16 volumio volumio[867]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
Mar 08 15:20:16 volumio volumio[867]: info: [1583680816053] CoreMusicLibrary::Adding element Media Servers
[... cut ...]
Mar 08 15:20:53 volumio volumio[867]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Mar 08 15:20:53 volumio sudo[1104]: pam_unix(sudo:session): session opened for user root by (uid=0)
Mar 08 15:20:53 volumio sudo[1104]: pam_unix(sudo:session): session closed for user root
Mar 08 15:20:53 volumio sudo[1107]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/ifconfig wlan0
Mar 08 15:20:53 volumio sudo[1107]: pam_unix(sudo:session): session opened for user root by (uid=0)
Mar 08 15:20:53 volumio sudo[1107]: pam_unix(sudo:session): session closed for user root
`

dubaleeiro, 2020-03-13:
I have just tried again with a different SD-Card and fresh Volumio install: same issue :)