volumio/volumio-plugins #426: Why an openbox-session ?

domcars0, 2020-04-02:
Hi,
Thanks for this usefull plugin. Just a question. I would like to know why we need an openbox-session? In fact why do we need any window manager as the browser is launched fullscreen and then don't need to manage any window ?
I've removed the line
`openbox-session & `
from the /opt/volumiokiosk.sh script and it works very well.
Thanks.

gvolt, 2020-04-02:
I have tried it and on my Raspberry Pi touchscreen I have a black border which I wasn't able to remove setting Chromium flags `--start-maximized `and / or `--start-fullscreen`.

In order to remove the border I had to set the flag `--window-size=800,480` in /opt/volumiokiosk. This solution would have the disadvantage that we would need to determine the resolution of the connected screen before Chromium is started. Maybe that's the reason for using a window manager.

As an alternative to openbox you can use matchbox-window-manager which does not launch pulseaudio and also allows to have a virtual keyboard (matchbox-keyboard).

gvolt, 2020-04-03:
Something like this could work:

```
#!/bin/bash
while true; do timeout 3 bash -c "</dev/tcp/127.0.0.1/3000" >/dev/null 2>&1 && break; done
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /data/volumiokiosk/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"None"/' /data/volumiokiosk/Default/Preferences
x=$(xrandr | grep \* | cut -d' ' -f4 | cut -d'x' -f1)
y=$(xrandr | grep \* | cut -d' ' -f4 | cut -d'x' -f2)
while true; do
  /usr/bin/chromium-browser \
    --disable-pinch \
    --kiosk \
    --no-first-run \
    --disable-3d-apis \
    --disable-breakpad \
    --disable-crash-reporter \
    --disable-infobars \
    --disable-session-crashed-bubble \
    --disable-translate \
    --user-data-dir='/data/volumiokiosk' 	--no-sandbox     http://localhost:3000 \
    --window-position=0,0 \
    --window-size=${x},${y}
done
```

gvolt, 2020-04-11:
@volumio 
I think with the above solution we could actually do without openbox. Do you have any objections to remove openbox in the upcoming next version of the touch display plugin?

volumio, 2020-04-13:
IMHO it could be a good idea. Can you send a PR for testing?