volumio/volumio-plugins #428: Touch Display Plugin - EACCESS

jjproe, 2020-04-13:
I have following error:

error: Error setting display brightness: Error: EACCES: permission denied, open '/sys/devices/platform/rpi_backlight/backlight/rpi_backlight/brightness'

gvolt, 2020-04-13:
Thanks for reporting.

Please provide some more information:
1. What is your board (Pi3B, Pi3B+...)?
2. What is the brand and model of the screen you have connected?
3. How is the screen connected (DSI, HDMI...)?
4. What is your Volumio version?
5. What version of the plugin are you running?
6. When/in what situation does the error occur?

RasPiPkr, 2020-05-09:
Hi gvolt,
When I shutdown I see a red message about backlight popup in top right corner, if that's normal then fine.
I also have a issue with touch screen plugin, is a onscreen keyboard as default as I don't get anything if I click on search box.
I have installed matchbox-keyboard as a result but have a issue with it only works once as the next time keyboard wont stay onscreen.

In regards to your information questions above:
1 Raspberry Pi 3B+
2 Official 7" Raspberry Pi Touchscreen
3 DSI
4 same thing happens on versions 2.714, 2.729 and latest 2.773
5 1.1.7
6 Keyboard works and search works but if I try a 2nd search keyboard flickers on screen for a split second and also so does the settings menu.
7 lists of files:
volumio@volumio:~$ ls -al /sys/devices/platform/rpi_backlight/backlight/rpi_backlight
total 0
drwxr-xr-x 3 root root    0 May  9 12:01 .
drwxr-xr-x 3 root root    0 May  9 11:41 ..
-r--r--r-- 1 root root 4096 May  9 12:01 actual_brightness
-rw-r--r-- 1 root root 4096 May  9 12:01 bl_power
-rw-r--r-- 1 root root 4096 May  9 11:41 brightness
lrwxrwxrwx 1 root root    0 May  9 12:01 device -> ../../../rpi_backlight
-r--r--r-- 1 root root 4096 May  9 11:41 max_brightness
drwxr-xr-x 2 root root    0 May  9 12:01 power
lrwxrwxrwx 1 root root    0 May  9 11:41 subsystem -> ../../../../../class/backlight
-r--r--r-- 1 root root 4096 May  9 11:41 type
-rw-r--r-- 1 root root 4096 May  9 11:41 uevent

I hope you can help me out.

gvolt, 2020-05-09:
Hm, permissions on /sys/devices/platform/rpi_backlight/backlight/rpi_backlight/brightness seem to be not ok. Execute `sudo /bin/chmod 0666 /sys/devices/platform/rpi_backlight/backlight/rpi_backlight/brightness` and your problem re. the "red message" should be solved. Adjusting the permissions should have happened on first install of the plugin though...

Matchbox-keyboard is another thing. Could post the content of /opt/volumiokiosk.sh? If you should be a member of the Volumio forum you can also PM me (member gvolt) there.